ros-navigation/navigation2 #3: Collection of existing solution

artivis, 2018-06-08:
It could be interesting to start inventorying existing navigation-related solutions to draws inspiration, lessons etc from. 
How about a `design/community_package` folder with per-topic collection ??

Here is a first few 

Map representation :  
- [grid_map](https://github.com/ethz-asl/grid_map)

Localization:
- [robot_localization](https://github.com/cra-ros-pkg/robot_localization)

Framework:
- [move_base_flex](http://wiki.ros.org/move_base_flex)

mkhansen-intel, 2018-06-11:
Thanks @artivis, we will take a look at all of these links.

mikeferguson, 2018-06-11:
I would add a few additional notes:

 * https://github.com/locusrobotics/robot_navigation is work in the similar direction as move_base_flex. They have some similarities, they also have some differences.

It would also be useful to look at the different global/local planners out there. This is probably the area in which the community has done the most work, and it would be useful to pay attention to any shortcomings in the planner APIs that they noted during such work.

moriarty, 2018-07-11:
@reinzor [asked](https://groups.google.com/forum/#!topic/ros-sig-navigation/UjPsW9DKpFY) a related question on the ros-sig-navigation, listing dwb & move_base_flex.

Rein are there any additions?

reinzor, 2018-07-12:
I think most relevant (non navigation stack 1) packages have been mentioned.

mkhansen-intel, 2018-08-03:
We've now looked at all of those packages listed, and have updated the repo with our Architecture concept, which should allow for using grid_maps and other algorithms. Take a look at this presentation for more info. [Navigation 2 Overview](https://github.com/ros-planning/navigation2/blob/master/doc/design/Navigation_2_Overview.pdf)

reinzor, 2018-08-16:
Hi @mkhansen-intel , thanks for the presentation! I am very interested in the design of the mentioned "Environmental model", at our university, we have also been working on models to represent the world. We developed the Environment descriptor (ED) https://github.com/tue-robotics/ed as a result of a PhD, graduation and Robocup@Home work. We use it to describe the world in an entity way that have various properties such as geometry, classes etc. It exposes plugin and service interfaces to query the world model and interact with it using perception plugins (adding semantic labels to detections), external python state-machines (querying object position and other geometric information such as areas), navigation (publishing an occupancy grid to be compatible with the nav stack 1.0), moveit planning scene etc. The current ED implementation is rather limited to a specific sensor set: RGBD camera's and Laserscanners, however this can be extended to other sensor ofcourse. It has been a development project for many years so don't expect final production code but it might be interesting to just look through the code and used concepts.

Some background (if interested) reading material https://bit.ly/2OChm3N (describes ED implementation) and some follow-up research regarding design of world models https://bit.ly/2OGzHwp (example implementations closed-source), more focusing on probablistic state estimation and tracking.

mkhansen-intel, 2018-08-30:
@reinzor - Thanks for pointing us to your project. We'll take a look at it this week.

mkhansen-intel, 2018-09-07:
Closing this issue as there's no work to be done on this at this time.

closed 2018-09-07 by mkhansen-intel