ros-navigation/navigation2 #6: Which license to use? BSD 3 Clause or Apache 2?

mkhansen-intel, 2018-07-03:
We need to resolve the licensing. ROS Navigation uses BSD 3. Most of ROS2 core uses Apache 2.0. For new code, we could license it Apache 2.0, but we assume there will be some code ported from ROS (BSD). How is this being handled by other ROS2 projects?

crdelsey, 2018-08-06:
I don't think we have any options here. If the file is BSD licensed, it must stay BSD licensed unless we replace all of its contents. The license says:

`
Redistributions of source code must retain the above copyright
      notice, this list of conditions and the following disclaimer.
`

mkhansen-intel, 2018-08-06:
True. For ported code, we need to retain the license. For new code, I believe we can use Apache 2.0. What I don't know is whether to put a LICENSE file in the repo or not since it will be a mixture. I'll get that clarified.

mkhansen-intel, 2018-09-17:
We have mixed Licenses, all new code will be Apache 2.0, and we will not have a LICENSE file in the repo due to the mixed licenses.  Closing this issue.

closed 2018-09-17 by mkhansen-intel