ros-navigation/navigation2 #21: Port Costmaps to ROS2

orduno, 2018-08-10:


mkhansen-intel, 2018-09-05:
As part of this work, the existing costmap placeholder code should either be removed or replaced. 
src/libs/nav2_util/src/costmap.cpp

bpwilcox, 2018-09-14:
We can port in smaller chunks from the existing ROS1 costmap_2d code in #86

SteveMacenski, 2018-09-24:
As part of this work, it might be good to keep in the back of your mind how you'd want to remove having 2 costmaps (one for controller another for planner) as that duplication of work accounts for a good amount of resource usage when using plugins for the 3D world.

mjeronimo, 2018-09-25:
@SteveMacenski  Yes, definitely. We plan to factor out the duplication, were possible.

mkhansen-intel, 2018-10-29:
Still need to port voxel layer (#193) and obstacle layer (#192), those are covered in their own issues, so closing this issue.

closed 2018-10-29 by mkhansen-intel