ros-navigation/navigation2 #35: Repo formatting

SteveMacenski, 2018-08-30:
Currently the metapackage exists under 'src' directory, the navigation stack and most other ROS metapackages are not organized this way and leads to a level of confusion

mkhansen-intel, 2018-09-25:
@SteveMacenski  - sorry for the late update on this one. I agree. I believe we could get rid of the top level 'src' folder and move all the sub-folders up to the top folder level. We also currently don't have a meta-package yet. We need one. I could work on a PR for these changes. @mjeronimo - do you have any issues with this request?

SteveMacenski, 2018-09-25:
For how things are generally organized, I think you could just move everything out of the src folder alongside the `tools` and `docs` directories for now and update the paths as needed. 

I could do this tomorrow or on the plane to ROSCon the day after. I think it might be a good way for me to jump into the project and get actively involved.

mjeronimo, 2018-09-26:
@SteveMacenski  What's the benefit of removing the src directory from the hierarchy and pushing the contained directories up one level?

SteveMacenski, 2018-09-26:
@mjeronimo The src directory doesn't follow the metapackage conventions in any other ROS repository (a number of other ROS2 examples linked below), it is confusing and what ends up happening is grouping together of independent packages within a bunch of subdirectories. If you look at the existing navigation stack, all the packages are in the root directory so they're easy to access. At the end of the day, these packages are not related to each other and frequently for professional users we don't utilize a vast majority of them. The navigation stack doesn't ship as 1 project, but as a collection of projects that aren't denoted by `localization` or `mission_execution`.

Metapackages should have all of their packages available in the root of the repository. Catkin and Colcon will search for them inside of non-package directories but there's no precedent for putting them in a source directory. A repository should not itself be a workspace and when its places inside src, it seems like you're utilizing this repository as itself a workspace and not that this metapackage is inserted into an existing workspace. 

There are a number of situations where you want to have multiple projects in the same workspace that do not depend on each other to grab rosdeps. (i.e. I'm shipping my robot with a couple of subsystems in different packages/repositories that don't depend on each other in 1 container I'm building together).  

Hope that makes sense.  

tl;dr
- Confusing grouping and these packages should be regarded as independent of each other
- src implies this repository is being used as a workspace, which shouldn't be happening
- There is plenty of precedence for metapackages and organization and this does not follow

https://github.com/ros-perception/vision_opencv/tree/ros2
https://github.com/ros-simulation/gazebo_ros_pkgs/tree/ros2
https://github.com/ros2/common_interfaces/tree/bouncy
https://github.com/ros2/turtlebot2_demo/tree/bouncy

mjeronimo, 2018-09-26:
@SteveMacenski  Thanks for the explanation and the references. Let's proceed and organize as you suggest.

SteveMacenski, 2018-09-27:
Ok, will PR this with updated docs and build scripts.

closed 2018-10-19 by SteveMacenski