ros-navigation/navigation2 #36: High level block diagram of intended workflow

SteveMacenski, 2018-08-30:
Coming from the standard nav stack and having built a few alternatives over time, I dont look at this project and have a good understand what's happening or how the elements interact.

Is there a plan to make a block diagram much like the move_base one to clear up your architectural intentions? It's challenging to get involved or track without it.

SteveMacenski, 2018-08-30:
I see this guy (https://github.com/ros-planning/navigation2/blob/master/doc/design/Navigation_2_Overview.pdf) but that's not really a full block diagram of what you're trying to accomplish here with navigation2 restructuring instead of starting with navigation stack and building on top of it

mkhansen-intel, 2018-08-30:
@SteveMacenski - Steve I've added this to our TO DO. We'll create a better diagram or two and put it in a more obvious place, like the top-level README. Thanks for your feedback.

SteveMacenski, 2018-08-30:
More generally speaking, I'd like to be involved in this effort as well. Having a good briefing about the thought process and what's being rebuilt vs. ported over would be good!

closed 2018-10-29 by SteveMacenski