ros-navigation/navigation2 #40: Test Navigation integration with robot_localization package (Kalman Filter)

mkhansen-intel, 2018-08-31:
We need to verify that our localization can work with the robot_localization package which has recently been ported to ROS2.

mkhansen-intel, 2018-08-31:
Link:
https://github.com/cra-ros-pkg/robot_localization/issues/265

mkhansen-intel, 2019-02-15:
@mhpanah - Do we still want to do this or should I close it?

mhpanah, 2019-02-19:
> @mhpanah - Do we still want to do this or should I close it?

We haven't done this yet. It's a good idea to do it at some point. However, low priority at the moment.

mkhansen-intel, 2019-09-17:
@rotu - I see you have been using the robot_localization package as part of your Rover demo. Are there any changes to Nav2 required or can I close this issue?

If it's working, maybe you could give us a quick overview of how you connected the two? Are you using AMCL?

rotu, 2019-09-17:
There are no changes to Nav2 required, but robot_localization has no current official branch working under Dashing, so I'm using an unofficial fork https://github.com/RoverRobotics-forks/robot_localization. @klintan was also working on the port to Dashing but that seems to have stalled.

Yes, I'm using it with AMCL.

klintan, 2019-09-17:
> There are no changes to Nav2 required, but robot_localization has no current official branch working under Dashing, so I'm using an unofficial fork [...]

See my answer here :) 
https://github.com/cra-ros-pkg/robot_localization/issues/473

SteveMacenski, 2019-09-17:
I'm still 2 layers deep getting everything in order so we can meaningfully talk about getting R_L shipped out. The state of the ROS2 port of R_L also isn't awesome so I have plans post-ROSCon to probably sit down and re-port it from melodic.

mkhansen-intel, 2019-09-18:
@rotu - can you run rqt_graph and show how you have connected the inputs / outputs of the robot_localization package to Nav2?

mkhansen-intel, 2019-10-02:
@rotu - I'm still interested in knowing how you got this to work, would you be willing to run rqt_graph on your running system and post back here? Even better would be a tutorial document. Do you have any bandwidth to help at all on this?

rotu, 2019-10-02:
Oops! I'm sorry I dropped the ball and thanks for the reminder. The node is called "/se_node" in my graph.

[image]

rotu, 2019-10-02:
And all my config lives here:

https://github.com/RoverRobotics/openrover-ros2/blob/master/openrover_demo/launch/presence.launch.py
https://github.com/RoverRobotics/openrover-ros2/blob/master/openrover_demo/config/presence.yaml

I'm not sure what tutorial would be helpful. It's just figuring out the parameters and topic mappings and such.

mkhansen-intel, 2019-10-03:
I'm just not clear on what purpose AMCL is serving vs the se_node. It looks like the map->odom /tf and the /odom topic are both being published by the se_node. If so, then is AMCL even needed?

SteveMacenski, 2019-10-03:
Without running it, it looks right to me with [world_frame](https://github.com/RoverRobotics/openrover-ros2/blob/master/openrover_demo/config/presence.yaml#L72) set to `odom`, its going to publish the odom-> base_link transform.

rotu, 2019-10-03:
se_node has no absolute positioning info and doesn’t know about the map frame. `/odom_raw` has only velocity info. That also means that, in theory, the se_node positional uncertainty should grow forever, but that doesn’t matter since we’re getting the absolute position from amcl instead.

mkhansen-intel, 2019-10-04:
Based on this, I think the intention of this issue, which was to prove someone can use robot_localization with nav2, has been met. I'm closing this issue!

SteveMacenski, 2019-10-04:
I’d like to keep this open until there’s an official release of RL for ROS2 that we can validate against. Right now its various folks’ forks which don’t necessarily represent a stable state of RL stable with Nav2 stable

mkhansen-intel, 2019-10-07:
Fair enough, I just made you the owner of this ticket as it met my original intent already

shivaang12, 2019-11-20:
Is someone working on this?

SteveMacenski, 2019-11-20:
Slowly, there’s alot of long standing background tasks that move at glacial speeds. I just finally got all the ROS2 PRs merged into mainline after possibly the longest code review in history (thanks @klintan for sticking with me). 

Next steps are to look at Rover’s fork and see if there’s anything worth adding or just different, thoroughly test, and deploy with bloom.

SteveMacenski, 2020-01-13:
As an update, my PR has been merged and at a high level tested. Mabel has another PR to add the missing stuff from Melodic over to Dashing (think estimator, etc). Once that is in, we can probably do a release. 

We are ready, though, to test the EKF and the UKF in Dashing against Navigation2. The basic stuff is there and in mainline for this work to actually take place now.

SteveMacenski, 2020-01-22:
Shown to work as part of the ROS2 porting

closed 2020-01-22 by SteveMacenski