ros-navigation/navigation2 #50: How does AMCL sensing ignores dynamic objects that has not been localized yet?

mhpanah, 2018-09-05:
Go over the code to see if it has functionality to ignore dynamic objects.

mikeferguson, 2018-09-11:
Not sure how far into this you are -- but it really doesn't ignore dynamic obstacles. Basically, the assumption is that the fixed obstacles largely outweigh the dynamic ones and so it tends towards the right answer. The implementation is about 20 years old, and follows the Probabilistic Robotics book pretty closely (the parameter descriptions in the ROS driver actually reference the book rather than describe in extreme detail how the parameters work)

SteveMacenski, 2018-09-11:
There's a beam skipping feature which attempts to do some of that - but I can't speak much to its effectiveness though seems to help some.

mikeferguson, 2018-09-11:
@SteveMacenski -- that's a good point -- I actually forgot that existed.

mhpanah, 2018-09-11:
@mikeferguson  that assumption make sense when the number of dynamic objects surrounding the robot is low. So far I've been just straight porting the amcl package and did some minor code refactoring.  I'm planning to see where amcl can be improved and omitting the dynamic objects could be one area for improvement. 

@SteveMacenski  I haven't yet looked at the beam skipping feature, but that seems promising. 

My goal is to improve amcl for our navigation package please let me know if you guys have any suggestions or if there are any features that would be nice to have.  Thanks in advance.

mhpanah, 2018-10-19:
@SteveMacenski @mikeferguson Do you guys know why in AMCL's motion model there is ODOM_MODEL_DIFF and ODOM_MODEL_DIFF_CORRECTED? Also OMNI and OMNI_CORRECTED?  Thanks.

Also, is the omni motion model also derived from Probabilistic Robotics textbook?

SteveMacenski, 2018-10-20:
In agreement with @mikeferguson, I went through a AMCL sprint a few months back and came to the same conclusion - and got a nice performance boost to boot.

mhpanah, 2018-10-22:
Thanks @mikeferguson and @SteveMacenski .
I will take a look at git-blame for amcl and then drop the old code.

@SteveMacenski It would be great if you could send me the copy of the derivation so that I can double check the code to make sure it's correct.

SteveMacenski, 2018-10-22:
I'll take a look when I get home later. But this should provide some details. https://answers.ros.org/question/54467/is-amcls-implementation-of-the-odometry-model-correct/

closed 2018-11-10 by mhpanah