ros-navigation/navigation2 #57: Populate README files for each package

mkhansen-intel, 2018-09-05:
README files are empty, these should be populated with basic descriptions of each package.

mkhansen-intel, 2018-09-25:
Current list of empty README files:
$ for file in $files; do if [ -s $file ]; then true; else echo $file; fi; done
./src/world_model/nav2_costmap_world_model/README.md
./src/libs/nav2_tasks/README.md
./src/libs/nav2_robot/README.md
./src/libs/nav2_util/README.md
./src/libs/nav2_libs_msgs/README.md
./src/navigation/nav2_bt_navigator/README.md
./src/navigation/nav2_simple_navigator/README.md
./src/control/nav2_dwa_controller/README.md
./src/planning/nav2_planning_msgs/README.md
./src/planning/nav2_astar_planner/README.md
./src/planning/nav2_dijkstra_planner/README.md
./src/mission_execution/nav2_mission_execution_msgs/README.md
./src/mission_execution/nav2_mission_executor/README.md
./src/mapping/nav2_map_server/README.md
./src/localization/nav2_amcl/README.md

Should we just remove these files?

mjeronimo, 2018-09-26:
No, I'd rather have the directory structure and files to be completed intact. I can handle populating the READMEs.