ros-navigation/navigation2 #59: Fix RCLCPP_INFO messages

mkhansen-intel, 2018-09-05:
Right now all the RCLCPP_INFO messages look like this one for example:
RCLCPP_INFO(get_logger(), "MissionExecutor::~MissionExecutor")

Let's change these to something more useful to the end user. For example, this one instead could be "Shutting down Mission Executor".

mjeronimo, 2018-10-05:
We've got lots of messages in the code that are for ourselves (developers working on the application) or other developers using this code. Is this suggesting that all of our messages should be for a non-developer end-user?  The traces reference are for the initial stages of development where we want to ensure that everything is created and shutdown properly.

mkhansen-intel, 2018-10-08:
For debug only messages, we can change them to RCLCPP_DEBUG() calls. See here:
https://github.com/ros2/ros2/wiki/Logging

Everything else (INFO, WARN, ERROR) should provide clear messages to the user.