ros-navigation/navigation2 #63: Strategy to resolve build dependencies

mkhansen-intel, 2018-09-05:
We have several build dependencies right now, like message_filters, that aren't available as packages. We need a method to track and then automate the pull and build of those if possible. Something like a build script that pulls the source, sources the correct setup files and runs colcon build.

crdelsey, 2018-09-13:
Added PR #83 that deals with this

mkhansen-intel, 2018-09-25:
**Proposed strategy**
For general library dependencies, such as a math library or tool, if it is a debian package:
1. Update the Dockerfile to apt-get install the library
2. Run the docker build to make sure that it works
3. Update the build instructions in the [BUILD.md](https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md) file with the 'apt-get install' command needed

For ROS2 source code dependencies:
1. Update the [ros2_dependencies.repos](https://github.com/ros-planning/navigation2/blob/master/tools/ros2_dependencies.repos) file with the correct repo and branch to pull the code from
2. Run the docker build to make sure the repos are pulled and built correctly by the buildscript

SteveMacenski, 2018-10-19:
@mkhansen-intel @crdelsey has this not been resolved?