ros-navigation/navigation2 #67: AMCL - fix all warnings

mkhansen-intel, 2018-09-07:
Currently when building AMCL, there are a lot of warnings spewing out. I realize that this code will be refactored, but want an issue to make sure these are cleaned up. I want to be able to add the -Werr compiler flag to the CMakeLists.txt going forward, so I'll file a separate issue for that.

mkhansen-intel, 2018-09-07:
Here is the build output with the warnings:
`
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/map/map.c: In function ‘map_get_cell’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/map/map.c:70:67: warning: unused parameter ‘oa’ [-Wunused-parameter]
 map_cell_t *map_get_cell(map_t *map, double ox, double oy, double oa)
                                                                   ^~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp: In member function ‘virtual bool amcl::AMCLSensor::UpdateAction(pf_t*, amcl::AMCLSensorData*)’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:48:37: warning: unused parameter ‘pf’ [-Wunused-parameter]
 bool AMCLSensor::UpdateAction(pf_t *pf, AMCLSensorData *data)
                                     ^~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:48:57: warning: unused parameter ‘data’ [-Wunused-parameter]
 bool AMCLSensor::UpdateAction(pf_t *pf, AMCLSensorData *data)
                                                         ^~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp: In member function ‘virtual bool amcl::AMCLSensor::InitSensor(pf_t*, amcl::AMCLSensorData*)’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:56:35: warning: unused parameter ‘pf’ [-Wunused-parameter]
 bool AMCLSensor::InitSensor(pf_t *pf, AMCLSensorData *data)
                                   ^~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:56:55: warning: unused parameter ‘data’ [-Wunused-parameter]
 bool AMCLSensor::InitSensor(pf_t *pf, AMCLSensorData *data)
                                                       ^~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp: In member function ‘virtual bool amcl::AMCLSensor::UpdateSensor(pf_t*, amcl::AMCLSensorData*)’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:64:37: warning: unused parameter ‘pf’ [-Wunused-parameter]
 bool AMCLSensor::UpdateSensor(pf_t *pf, AMCLSensorData *data)
                                     ^~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/sensors/amcl_sensor.cpp:64:57: warning: unused parameter ‘data’ [-Wunused-parameter]
 bool AMCLSensor::UpdateSensor(pf_t *pf, AMCLSensorData *data)
                                                         ^~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/map/map_draw.c:158:0: warning: ISO C forbids an empty translation unit [-Wpedantic]
 #endif
 
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf.c: In function ‘pf_update_converged’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf.c:219:10: warning: unused variable ‘total’ [-Wunused-variable]
   double total;
          ^~~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf.c: In function ‘pf_cluster_stats’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf.c:473:29: warning: unused parameter ‘pf’ [-Wunused-parameter]
 void pf_cluster_stats(pf_t *pf, pf_sample_set_t *set)
                             ^~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf_kdtree.c: In function ‘pf_kdtree_equal’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf_kdtree.c:186:34: warning: unused parameter ‘self’ [-Wunused-parameter]
 int pf_kdtree_equal(pf_kdtree_t *self, int key_a[], int key_b[])
                                  ^~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl/pf/pf_draw.c:163:0: warning: ISO C forbids an empty translation unit [-Wpedantic]
 #endif
 
In file included from /home/mkhansen/ros2_dev/ros2_ws/install/tf2/include/tf2/utils.h:20:0,
                 from /home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl_node.cpp:30:
/home/mkhansen/ros2_dev/ros2_ws/install/tf2/include/tf2/impl/utils.h:114:2: warning: extra ‘;’ [-Wpedantic]
 };
  ^
/home/mkhansen/ros2_dev/ros2_ws/install/tf2/include/tf2/impl/utils.h:148:2: warning: extra ‘;’ [-Wpedantic]
 };
  ^
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl_node.cpp: In member function ‘void AmclNode::globalLocalizationCallback(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<std_srvs::srv::Empty_Request_<std::allocator<void> > >, std::shared_ptr<std_srvs::srv::Empty_Response_<std::allocator<void> > >)’:
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl_node.cpp:882:78: warning: unused parameter ‘request_header’ [-Wunused-parameter]
[... cut ...]
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl_node.cpp:913:66: warning: unused parameter ‘request_header’ [-Wunused-parameter]
 AmclNode::setMapCallback(const std::shared_ptr<rmw_request_id_t> request_header,
                                                                  ^~~~~~~~~~~~~~
/home/mkhansen/ros2_dev/ros2_ws/src/ros-planning/navigation2/src/nav2_localization/amcl/src/amcl_node.cpp:915:70: warning: unused parameter ‘res’ [-Wunused-parameter]
                     std::shared_ptr<nav_msgs::srv::SetMap::Response> res)
`

mkhansen-intel, 2018-09-25:
@mhpanah - I think this is done, can we close?

mhpanah, 2018-09-26:
@mkhansen-intel Yes. It's done.

closed 2018-09-26 by mhpanah