ros-navigation/navigation2 #80: -Wpedantic flag

mhpanah, 2018-09-12:
What to do with -Wpedantic flag? Should we remove it?

crdelsey, 2018-09-13:
My recommendation is that we build with `-Wall -Wextra -Wpedantic -Werr` but get rid of -Wpedantic if it becomes a problem on specific components.

I think the escalation is:
1. Try to disable the warning using a pragma in the code
2. Add -Wno-pedantic in the narrowest scope of the CMakeLists.txt as possible
3. Remove -Wpedantic from the CXX_FLAGS for the package.

mkhansen-intel, 2018-09-14:
I agree with you Carl.

The immediate problem is AMCL right? I think those are fixable, right? At least the ones I see are extra semicolons after the curly braces, are there others that can't be fixed?

mhpanah, 2018-09-19:
Per Carl's suggestion I updated map_draw and  pf_draw with the following pragma:
#pragma GCC diagnostic ignored "-Wpedantic"

closed 2018-09-19 by crdelsey