ros-navigation/navigation2 #81: Find another yaml-cpp package

bpwilcox, 2018-09-13:
For compilation and execution, the nav2_ map_server requires the rviz_yaml_cpp_vendor package for parsing YAML files. We should find another yaml package that's not dependent on any rviz configurations.

lucbettaieb, 2018-09-13:
I'd like to open the discussion up to using a different "markup language" for map description files.  We could find a library that would work a lot better with the new stack.  JSON + PicoJSON is pretty good.

mikeferguson, 2018-09-13:
YAML is super pervasive in ROS1 - before changing away I would suggest checking what is pervasive in ROS2. For instance, nearly every config heavy package in ROS1 uses YAML to set parameter blocks in Launche files. I’d suggest that limiting the number of “languages” involved improves user experience.

SteveMacenski, 2018-09-13:
I agree with @mikeferguson. I'm all for json or something less horribly slow to load, but I'm not sure its worth breaking consistency across the ecosystem.

lucbettaieb, 2018-09-13:
I do get that, but.. when would be a better time than now? :grin:

SteveMacenski, 2018-09-13:
It should probably be part of a much large discussion that would have to involve the broader community and probably ROS2 development efforts at Open Robotics.

If the line in the sand is the map_server you're probably not really going to get community traction and remain largely un-adopted. The move here would be to start a much larger discussion (and hey, ROSCon is right around the corner) and get the relevant stakeholders buy-in (Open Robotics, maintainers, these Intel folks, and maintainers of ROS1 packages in critical path needed to be ported) as the burden will fall onto them.

crdelsey, 2018-09-14:
@SteveMacenski @lucbettaieb Is there really that much of a performance difference between YAML and JSON? I thought YAML was basically the same except with whitespace for structure instead of brackets. Is performance the only reason to want to replace YAML with something else?

SteveMacenski, 2018-09-14:
I'd encourage you to test for yourself with python json and yaml libraries (even using the Cloader options). Grab some large file and compare parse times. Yaml is much slower

SteveMacenski, 2018-10-26:
@bpwilcox given we use `yaml_cpp_vendor` now in map server, is this ticket done?

mkhansen-intel, 2018-10-29:
Closing since we use yaml_cpp_vendor now

closed 2018-10-29 by mkhansen-intel