ros-navigation/navigation2 #99: Integrate into ROS Build Farm for build, test and deployment

mkhansen-intel, 2018-09-25:
Need to integrate into https://ci.ros2.org/ prior to Crystal release so our packages will be released.

See repo here for more info:
https://github.com/ros2/ci

Nickolaim, 2018-10-02:
I can work on this.  But I will need more information for understanding how it works.  For example, I looked at the last Linux job as of today - [https://ci.ros2.org/job/ci_linux/5246/](https://ci.ros2.org/job/ci_linux/5246/). It has a link to [a gist with repositories to be built](https://gist.github.com/wjwwood/dd841b416086fc52a11bf9bfbdac0f63). How do repositories get to this list? How to define a job for CI and more importantly how to test it?

However, if this process requires setting up a local Jenkins environment with clients on both Mac and Windows - I don't have them.

crdelsey, 2018-10-02:
@Nickolaim Thanks for investigating that. The gist you sent looks a lot like the repos file in https://github.com/ros2/ros2

Maybe it is a simple as submitting a PR to that repo with an update to the ros2.repos file there?

There is a [file](/ros-planning/navigation2/blob/master/tools/ros2_dependencies.repos) in our repo with several dependencies that would need to be added or addressed in the same commit. Specifically:
1. **ros2_message_filters** Needs to be added at the same time.
2. **gazebo_ros_pkgs** Is already part of ROS 2 so can be left out.
3. **angles** Needs to be added at the same time.
4. **ros/launch** Is already in ROS2 so can be left out.
5. **xmlrpcpp** I think we want to remove this dependency before adding our code to the official ROS 2 build.

SteveMacenski, 2018-10-19:
It seems pretty analog to ROS 1 build farm if you interact with Bloom/rosdistro at all (I maintain a few bloom packages). I'll take a look at this on the side as well but won't be a top priority for me.

mkhansen-intel, 2018-10-29:
I see they've moved the ROS2 documentation. Here's a link to the new instructions for integrating with the build farm:
https://index.ros.org/doc/ros2/Releasing-a-ROS-2-package-with-bloom/

SteveMacenski, 2018-10-29:
Awesome, I can take a look at it this week. I'll need to make a release repository. I can have on my account but may be useful at some point to have it on the ros-planning organization or at least give access to you all as well.

SteveMacenski, 2018-10-30:
I added your team to my release repository. https://github.com/SteveMacenski/navigation2-release

Will work on this later in the week

SteveMacenski, 2018-10-30:
https://github.com/ros-planning/navigation2/pull/261

this is needed to get all the package versions the same to release via bloom

SteveMacenski, 2018-10-30:
(Notes to self)

1 add to index on https://github.com/ros2/rosdistro (with and immediately do bloom OR (do this) without, then on bloom update and add all to PR)

2 https://github.com/ros-planning/navigation2   https://github.com/SteveMacenski/navigation2-release
4.1 export ROSDISTRO_INDEX_URL='https://raw.githubusercontent.com/ros2/rosdistro/ros2/index.yaml'
4 bloom-release --rosdistro bouncy --track bouncy navigation2 --edit
5 add icon on readme for CI ros build farm on top of Travis's

```
  navigation2:
    doc:
      type: git
      url: https://github.com/ros-planning/navigation2.git
      version: master
    release:
      packages:
      - costmap_queue
      - dwb_core
      - dwb_critics
      - dwb_msgs
      - dwb_plugins
      - nav2_amcl
      - nav2_astar_planner
      - nav2_bringup
      - nav2_bt_navigator
      - nav2_controller_dwb
      - nav2_controller
      - nav2_costmap_2d
      - nav2_costmap_world_model
      - nav2_dijkstra_planner
      - nav2_dynamic_params
      - nav2_map_server
      - nav2_mission_executor
      - nav2_msgs
      - nav2_robot
      - nav2_simple_navigator
      - nav2_system_tests
      - nav2_tasks
      - nav2_util
      - nav2_voxel_grid
      - navigation2
      - nav_2d_msgs
      - nav_2d_utils
      tags:
        release: release/bouncy/{package}/{version}
      url: https://github.com/stevemacenski/navigation2-release.git
      version: 0.1.0
    source:
      type: git
      url: https://github.com/ros-planning/navigation2.git
      version: master
    status: maintained
```

SteveMacenski, 2018-10-30:
I ran bloom here's the PR for rosdistro. https://github.com/ros2/rosdistro/pull/283 I set up it to work with testing, docs, running jobs on PRs (can get rid of travis if we like). 

There were some unresolved dependencies (ie the packages we have in the ros2_depedencies) we need to add to their respective mainlines, or add keys for packages those people intend to maintain. 

I'll take care of the nav2_gazebo_localizer, nav2_planning_tests, roscpp.  I'm not actually sure how this compiles at all if `nav_core2` isn't here anymore. 

```
Could not resolve rosdep key 'message_filters'
Failed to resolve message_filters on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'message_filters', aborting.
message_filters is depended on by these packages: ['nav2_amcl', 'nav2_costmap_2d']
<== Failed
Could not resolve rosdep key 'nav2_gazebo_localizer'
Failed to resolve nav2_gazebo_localizer on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'nav2_gazebo_localizer', aborting.
nav2_gazebo_localizer is depended on by these packages: ['navigation2']
<== Failed
Could not resolve rosdep key 'nav2_planning_tests'
Failed to resolve nav2_planning_tests on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'nav2_planning_tests', aborting.
nav2_planning_tests is depended on by these packages: ['navigation2']
<== Failed
Could not resolve rosdep key 'nav_core2'
Failed to resolve nav_core2 on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'nav_core2', aborting.
nav_core2 is depended on by these packages: ['nav_2d_utils']
<== Failed
Could not resolve rosdep key 'roscpp'
Failed to resolve roscpp on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'roscpp', aborting.
roscpp is depended on by these packages: ['nav_2d_utils']
<== Failed
Could not resolve rosdep key 'tf2_sensor_msgs'
Failed to resolve tf2_sensor_msgs on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'tf2_sensor_msgs', aborting.
tf2_sensor_msgs is depended on by these packages: ['nav2_costmap_2d']
<== Failed
Could not resolve rosdep key 'xmlrpcpp'
Failed to resolve xmlrpcpp on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'xmlrpcpp', aborting.
xmlrpcpp is depended on by these packages: ['nav2_costmap_2d']
<== Failed
Could not resolve rosdep key 'yaml_cpp_vendor'
Failed to resolve yaml_cpp_vendor on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'yaml_cpp_vendor', aborting.
yaml_cpp_vendor is depended on by these packages: ['nav2_map_server']
```

SteveMacenski, 2018-10-30:
From the file 
```
repositories:
  ros2_message_filters:
    type: git
    url: https://github.com/intel/ros2_message_filters.git
    version: master
  gazebo_ros_pkgs:
    type: git
    url: https://github.com/ros-simulation/gazebo_ros_pkgs.git
    version: ros2
  angles:
    type: git
    url: https://github.com/ros/angles.git
    version: ros2
  ros2/launch:
    type: git
    url: https://github.com/crdelsey/launch.git
    version: TimerActionWorkaround
  xmlrpcpp:
    type: git
    url: https://github.com/bpwilcox/xmlrpcpp.git
version: master
``` 
looks like this conversion involves @bpwilcox  @crdelsey Intel's ROS2 message filters, angles, and gazebo ros pkgs

SteveMacenski, 2018-10-30:
It also looks like gazebo_ros_pkgs and angles are already in rosdep. Why are they in the ros2_dependencies file rather than using wstool?

https://github.com/ros2/rosdistro/blob/ros2/bouncy/distribution.yaml#L467

SteveMacenski, 2018-10-31:
we're now released as 0.1.0 in bouncy. However, since these dependencies are not resolved, the CI build farm trials will fail. We need a path to getting the ros2_dependencies as rosdep keys, merged into mainline, or as a rosinstall file (I think it'll take that)

SteveMacenski, 2018-11-12:
@mkhansen-intel  There's still more work to be done after those are resolved. Those are the current blockers to completing the rest. 

I wouldn't close this ticket until we have a successfully building CI + bloom released shadow-fixed metapackage

crdelsey, 2019-01-22:
We made it into Crystal release. Closing this now.

closed 2019-01-22 by crdelsey