ros-navigation/navigation2 #101: Investigate stall in planning tests

mjeronimo, 2018-09-26:
There are occasional stalls in the planning test when the dijkstra planner requests the costmap from the test node. This appears to be in the ROS code, but requires further investigation.

mkhansen-intel, 2018-09-26:
I don't know enough about this to triage it. Is the stall causing a failure?  Should this be labeled as a bug or task?

mjeronimo, 2018-09-26:
It's not causing a failure. There is a delay, sometimes of a couple seconds, as the dijkstra planner is waiting for the costmap to return from the costmap service. I just ran the tests and it happened a couple times during a run of 1000 tests. I'm wondering what the cause is of this and this issue is a reminder to investigate.

mjeronimo, 2018-09-29:
Upon closer inspection, I see that the multicast traffic on the network when nodes are joining and leaving (as happens with the test code when test nodes are created and destroyed) impacts the timeliness of the processing of input messages.

closed 2018-09-29 by mjeronimo