ros-navigation/navigation2 #103: Support / maintenance plan for DWB Local Planner

mkhansen-intel, 2018-09-26:
Currently we're porting DWB into the ROS2 Navigation repo, with the goals of:

1. Getting local planning working in the short term with the best known algorithm in ROS

1. Making it easy to download, build and test the Navigation stack

However we are essentially forking @Dlu's code. This could lead to maintenance headaches as he updates and moves his repo forward over time.

So what are we going to do? 

1. One option is to submit our changes to David's upstream repo, and just pull in the package as a dependency. 

1. Another option is to maintain this as a reference, with directions on how to replace it if the user wants to. 

Thoughts are welcome.

mikeferguson, 2018-10-22:
I'd actually suggest that if DWB is the new "standard/default" planner, that library might as well live here. I think the main reason @DLu didn't put the library in the existing ROS1 navigation repo was because he was targeting mainly his next-gen interface. If we can make the default library here support this repo's needs as well as his, it'd be really nice to have all the "default" planners exist within the repo, without much external dependencies.

mkhansen-intel, 2018-10-29:
This issue is now being addressed in #229

closed 2018-10-29 by mkhansen-intel