ros-navigation/navigation2 #110: Rename packages and folders along the lines of nav2_<type of package>_<specific_implementation>

crdelsey, 2018-09-27:
As we move to a flat directory structure, we can preserve the logical structure by naming packages according to where they would belong in the hierarchy

Instead of `nav2_dwa_controller` and `nav2_dwb_controller`, we should call them `nav2_controller_dwa` and `nav2_controller_dwb`

mkhansen-intel, 2018-10-29:
This won't be done. We're going to be consistent with existing package names like nav2_dijkstra_planner, etc.

closed 2018-10-29 by mkhansen-intel

SteveMacenski, 2018-10-29:
@mkhansen-intel I do vaguely like this for the planners and navigators as there may be a number of them and it would naturally group them in alphabetical order in the repo. Basically just changing `nav2_dijkstra_planner` -> `nav2_planner_dijkstra` and the navigators similarly. 

Thoughts?