ros-navigation/navigation2 #111: Create a class diagram that shows the relationship between packages in the repo

crdelsey, 2018-09-27:
To help those new to the source base, a class diagram that shows the logical structure of the different packages would be helpful.

shivaang12, 2019-10-31:
@crdelsey UML class diagram would be okay? or you mean something else?

yathartha3, 2019-10-31:
We have done UML before, and I think it is a good idea. PlantUML is what we used to before generate the diagrams from vscode.

shivaang12, 2019-10-31:
@yathartha3 Are you making working on it?

yathartha3, 2019-10-31:
@shivaang12 It was one of the things I was considering to do since I was starting to look at the documentation. However I haven't started working on it, and if you are interested, go ahead.

shivaang12, 2019-10-31:
@yathartha3 I can do it if that's okay.

SteveMacenski, 2019-11-01:
I disagree, I think this should be something easier for a random person to understand. Not everyone (maybe even most?) that works in navigation/robotics have formal computer science backgrounds & this type of document should be accessible to anyone that knows ROS. Lots of software beginners use navigation and we shouldn’t bog a high level “how everything talks” down with notation. 

I’m thinking this would looks like a nicer version of the classic move base diagram

shivaang12, 2019-11-01:
@SteveMacenski Do you mean a higher level diagram than normal UML class diagrams?

SteveMacenski, 2019-11-01:
I’m saying a nice ROS-based diagram of data flow, no UML

crdelsey, 2019-11-04:
> I’m saying a nice ROS-based diagram of data flow, no UML

If you keep the diagram super simple and don't try to express any details, this can work, although, IMHO, a UML diagram is just as easy to understand in that case. In my experience, if people don't use UML, and want to express detail, they tend to invent their own notation which is even harder to understand because they alone in the world know what it means.

shivaang12, 2019-11-04:
Lets create both UML and simple data flow. UML for new contributors to understand our design and data flow for the users to understand.

SteveMacenski, 2019-11-05:
Fair enough on the both comment. 

I am referencing a “rqt” like version, arrows with topics/services/action names (maybe types). I think that notation is inherently understandable to everyone. This goes to that over /topic. And the names of /topic give a clear understanding of what’s being transported. If /topic doesnt, /topic should change

shivaang12, 2019-11-06:
All things consider, I am creating UML and rqt type diagram ( or whatever steve described).

SteveMacenski, 2019-11-07:
I’m basically just describing what Matt already made at ROSCon, with maybe a couple tweaks.

shivaang12, 2019-11-07:
Want to discussed it in nav_wg?

SteveMacenski, 2019-11-07:
I’m still in China. No wg this week for me.

shivaang12, 2019-11-07:
Okay, I will work on rqt type diagram. We'll discuss later about the other.

shivaang12, 2019-11-12:
@SteveMacenski I don't know if this aligns with what you were saying.
[image]

crdelsey, 2019-11-18:
This looks good, but is missing a few connections I think. /map should go to nav2_planner, and /scan should go to nav2_planner and nav2_controller.

It should be possible to use `rqt` to auto-generate the graph.

SteveMacenski, 2019-11-18:
I think its missing quite a bit. It doesn't actually show any structure behind the project. 

BT Nav connects to planner / controller / recovery and each of those load maps of plugins. 

Topics also as @crdelsey says. I don't think Lifecycle or map server should be given

shivaang12, 2019-11-18:
@crdelsey This is the rqt_graph. I agree that above diagram is missing quite a bit. Was trying to confirm if it aligns to what @SteveMacenski  was referring earlier as "rqt" type diagram. I think now I have a good picture about what to do.

[image]

SteveMacenski, 2019-11-18:
> I’m basically just describing what Matt already made at ROSCon, with maybe a couple tweaks.

@mkhansen-intel can you just post that slide for us?

SteveMacenski, 2019-11-19:
[image]

See above. This is what I’m thinking. Closer in line with how the move base diagram was laid out. A UML as well wouldn’t be bad, but that’s too in the weeds I think for a first (or even 10th) look at the stack. 

This would be color coded by optional inputs & all the connections would have labelled a interface type and topic name. 

I haven’t figured out how (or if) to represent the costmaps. The rectangles in the server boxes are the plugin maps of algorithms.

SteveMacenski, 2019-11-20:
[image]

thoughts? still in the works but... actually legible :rofl:

SteveMacenski, 2019-11-20:
Version with more information (I think I'm happy with this one, plus or minus color scheme)

[image]

mlherd, 2019-11-20:
Looks very good to me. What about odom?

mlherd, 2019-11-20:
this is I belive what Matt used for ROSCon.
[image]

SteveMacenski, 2019-11-20:
> What about odom?

What about it? its not really a substantial element. REP-105 provides me the transforms I need in TF that replaces the use of the odom topic in 95% of cases.

SteveMacenski, 2019-11-20:
> PS: using different colors for the arrows (topics, actions etc) and using "/" for the topics such as /map, /tf etc. can be helpful to the reader.

I had that on a list of comments as well. I decided against it since the message types I show include the `/msg/` and `/action/` but still not necessarily sold one way or another. Especially considering that everything here is an action except the arrows on the sides

SteveMacenski, 2019-11-20:
Didn't even think about this - but all the action servers have double sided arrows (except for waypoint follower, which I can and should change) so the key could just that to separate actions vs topics. 

If you also have thoughts on how to include the `ComputePathToPose` and `FollowPath` cleanly, I'd love to have those included but was alot of letters for a small space

mlherd, 2019-11-20:
Also the sensor data can be used by both controller server and planner server. + Is it worth mentioning the param file?

mlherd, 2019-11-20:
> If you also have thoughts on how to include the ComputePathToPose and FollowPath cleanly, I'd love to have those included but was alot of letters for a small space. 

Just an idea.

[image]

mlherd, 2019-11-20:
> What about it?

Since the /odom is used by DWB, I just wanted to know why you didn't include it. I aggree that the graph should be as generic as possible. Now, it makes sense why you didn't include map server and amcl to the picture.

mlherd, 2019-11-20:
> Other than colors, any other ideas? different types of arrows? Fonts?

I think the arrows should touch the shapes. In my opinion, that will make it easier to see where the arrows go and where exactly they come from. for example, the behaviour tree arrow should touch the bt navigator server shape, and cmd_vel should touch controller server etc
.

shivaang12, 2019-11-20:
I thought lifecycle manager should be included as it is the core thing in our stack. But this looks generic and useful for a newbie to understand the core operations. I agree with @mlherd about the arrows.

SteveMacenski, 2019-11-20:
Comments in order:

It is true, sensor data is in each costmap. Thoughts on how to make that look nice?

I can live with the actions key you suggest, good add! 

I think from looking at it the slight arrow offset from the box is clearly indicating what its pointing to in particular, but I also made it so I’m clearly biased. Anyone else feel its ambiguous? 

The lifecycle manager is just going through state transitions. That not really something I think people need to be aware of on first glance. Most users for toy examples, school projects, or algorithm designers could live without ever knowing its there. Also the plan long term is to have it part of the launch bringup so it will be going away relatively soon.

mlherd, 2019-11-20:
> It is true, sensor data is in each costmap. Thoughts on how to make that look nice?

[image]

SteveMacenski, 2019-11-20:
[image]

Intermediate. Still working out what I want to do with sensor data. I like having the "ins" and "outs" being on the same axis and the auxiliary things on another axis.

SteveMacenski, 2019-11-20:
[image]

Ok. Ready for another round of thoughts.

closed 2019-11-20 by SteveMacenski

mlherd, 2019-11-20:
It look cool on the main page :)
IMO, what the arrows that are between the planner server, the controller server and the recovery server mean is a little bit unclear in the diagram