ros-navigation/navigation2 #132: Port tf2_sensor_msgs

bpwilcox, 2018-10-02:
In the costmap_2d package, the observation_buffer requires the tf2_sensor_msgs package which is not yet ported geometry2 metapackage.

bpwilcox, 2018-10-23:
> Is this the port, or is there additional work required?
> https://github.com/ros2/geometry2/tree/ros2/tf2_sensor_msgs

It hasn't been ported yet and is currently being ignored (AMENT_IGNORE) by the compiler. It is a small package, however.

SteveMacenski, 2018-10-23:
https://github.com/ros-planning/navigation2/issues/192#issuecomment-432371318

mkhansen-intel, 2018-10-23:
See issue: https://github.com/ros2/geometry2/issues/59

SteveMacenski, 2018-10-23:
Ported: https://github.com/ros2/geometry2/pull/74

SteveMacenski, 2018-10-26:
done.

Once geometry2 merges my PR (its been approved) I'll remove my dep fork of it and then the ros2_ws geometry2 will contain it

closed 2018-10-26 by SteveMacenski