ros-navigation/navigation2 #135: Create a navigation_bringup package

mkhansen-intel, 2018-10-03:
We need a package for launching the fulll navigation stack. This will be the primary way users launch the stack as a whole. Most ROS packages have a bringup, so this is pretty standard. This is where the main launch file should live. 

The command line should be something like this:
ros2 launch navigation2_bringup nav2_bringup.launch.py

mkhansen-intel, 2018-10-03:
We can/should use Turtlebot3 as our default robot, but there shouldn’t be any hard coded dependencies on Turtlebot3 in the bringup launch. It should be a run-time parameter. Then the Turtlebot3 team for example will create a turtlebot3_navigation package which calls our navigation2_bringup with the correct parameters for TB3. For our testing, we can have navigation2_turtlebot3.launch.py or something like that in our system test directory.

SteveMacenski, 2018-10-08:
Will be added to this PR as needing to not be in root for this reorg https://github.com/ros-planning/navigation2/pull/107

closed 2018-10-10 by mkhansen-intel