ros-navigation/navigation2 #148: Repository decisions

SteveMacenski, 2018-10-08:
Hi,

Getting back in from Spain. I'm seeing alot of things that happened over IROS that concerns me. There's packages being split up with logical grouping and new packages that are simply[ empty "examples"](https://github.com/ros-planning/navigation2/tree/master/src/control/nav2_controller_example) in place of previous working and supported code. I think there needs to be more discussion before reorganizing things so dramatically. Moreover, removing existing code that a large number of people depend on (DWA). Simply removing these things in mass may make it impossible for a large number of organizations to ever move over to this. 

Its gotten to a point where its nearly impossible to contribute because things are being broken up in ways without a discussion and (at least) I can no longer understand the architectural decisions. Packages that have been working and plugin based are being broken out into many **many** packages or creation of a generic "utils" package that seems to just be dumping sections of projects creating a great deal of redundancy and removing configurability and modularity of those packages that ROS is famous for. 

I'd like this ticket to be a discussion place to help facilitate finding a better method so that these decisions can be made after getting some OKs from relevant end-user and developer stakeholder approval, even if it slows down development. As one person effected, I can't revert or argue for all the things I see happening that I'm deeply concerned about for long term maintainability and usability as I have a day job that takes priority. 

Some examples of concerns
- removing dwa with a non-working replace holder
- costmap 2d keeps moving around and broken out
- amcl being broken apart and placed in utils 
- repository organization and violating ROS conventions
- moving so quickly without any (public) end-user critical feedback
- this utils package with seemingly random things from existing working packages thrown into it, violating basic encapsulation that packages should contain

mkhansen-intel, 2018-10-10:
Hi @SteveMacenski,
Per our conversation yesterday, we appreciate the feedback and we're trying to address these issues. In general, some of the changes we're making are to refactor and re-organize things for better separation of concerns where possible. We're definitely open to feedback. Please file issues one at a time for things so that we can address them individually and we'll do that. I'd like to close this meta-ticket based on that and we can address issues in individual tickets. OK?

SteveMacenski, 2018-10-10:
Sounds good.

closed 2018-10-10 by SteveMacenski