ros-navigation/navigation2 #153: Pluginlib plugins for AMCL sensor models

SteveMacenski, 2018-10-09:
to swap out uses of different models. Examples in laser, create plugins for the beam, MLE, and MLE prob models so you can swap at run time, but then also have the infrastructure to swap out at run time with any customly chosen one implemented.

Same with the odometry models for corrected-* and diff, omni

mjeronimo, 2018-10-10:
I'd rather see us use a separate sensors package. That is, utilize ROS nodes for the various sensors instead of C++ plug-ins.

SteveMacenski, 2018-10-10:
AMCL is generally an embedded process for many users, it would allow people to shoot themselves in the foot by writing some of these things in python. I'm not too familiar with the overhead and delays associated with DDS. Would the transport time and decode be essentially analog to a shared pointer when in the same process?

SteveMacenski, 2018-12-17:
Sounds like the thought is to include in robot header not as plugins

closed 2018-12-17 by SteveMacenski