ros-navigation/navigation2 #161: Create Issue and PR formats / templates

SteveMacenski, 2018-10-11:
GitHub allows you to do this (https://github.com/ros-planning/navigation2/community) and its useful so when others want to file tickets or PRs, we can create a recommended format or reminders to include important information. 

Intel's librealsense2 repo has a good example of such a template (https://github.com/IntelRealSense/librealsense/issues/new)

crdelsey, 2018-10-30:
I don't see a  way to create a PR template, but I've created an issue template based on OSRF's template for RCLCPP.

It is pretty generic, but I figure we can add to it as we go along.

The Realsense one seemed a bit too specific to their stuff

SteveMacenski, 2018-10-30:
Definitely. Start with something and add more as we go. 

I can lob out a starter PR template then