ros-navigation/navigation2 #163: Robot class TODO list

mhpanah, 2018-10-12:
In Robot class, getCurrentVelocity method gets odometry msg which contains both twist and pose. getCurrentPose method gets pose msg which is estimated from localization module. To make this less confusing, rename the getCurrentVelocity method to getOdometry and also rename the getCurrentPose to getCurrentEstimatedPose.

Also, the following methods needs to be implemented in Robot class:
TODO(mhpanah):
-    getFootPrint()
-     robot types and SN or ID
-     Add unit and integration test
-     Be able to configure which topic name to subscribe to
-     Topic names for pose and odom should should be confifured with parameters

SteveMacenski, 2018-10-18:
This class also needs timeouts, I'm looking through it for another ticket and once you recieve poses you store and then dish out after an initialization bool is set. There should be a timeout on the info in case something happens and info stops publishing you can produce a warning that you can't transform. I think TF should be involved here for this reason.

mhpanah, 2018-10-19:
@SteveMacenski That's a good point, I agree. Initially, I was thinking to have the client to check the timestamp to see if the latest info is useful. However, it's safer to handle it in robot class in case the user is not checking the timestamp.  If the latest info is not within a pre-defined threshold the robot class can throw a warning and also provide the latest info that it has.

mkhansen-intel, 2019-02-15:
@mhpanah - is this done?

crdelsey, 2019-02-15:
@mkhansen-intel  No. Not done yet

mhpanah, 2019-02-19:
> @mkhansen-intel No. Not done yet

We should all brainstorm to see what functionalities we need in robot class.  It's a low priority at the moment.

closed 2019-07-01 by crdelsey