ros-navigation/navigation2 #171: re-enable Message Filter in AMCL

mhpanah, 2018-10-15:
currently geometry2 message_filter is not ported to ROS2.  Once the message filter is ported to ROS2 we need to re-enable it in AMCL.  

@gaoethan Would you be able to help us with this?

gaoethan, 2018-10-16:
Yes. generally speaking, we can help on this. when do you plan to use it ?

mkhansen-intel, 2018-10-16:
@gaoethan - we're currently using a work-around in the code to make AMCL work without it, so we could use this change as soon as you can make it. How long would it take?

gaoethan, 2018-10-18:
Okay, I think that it's enough for me to port it for you in one week.

mkhansen-intel, 2018-10-18:
@gaoethan - 2 weeks is fine if you have other work that is higher priority

yechun1, 2018-10-18:
Seems the message_filter interface of tf2_ros from geometry2 is not ported, let me have a try to port it.

mkhansen-intel, 2018-10-23:
Also linking to our issue on this #132

gaoethan, 2018-11-08:
I've already worked out an available one, please have a try with the following code base:
https://github.com/gaoethan/message_filters.git with branch `master`
https://github.com/gaoethan/geometry2.git with branch `tf2_ros_message_filters`

mkhansen-intel, 2018-11-15:
When we enable the tf2 message filters, we need to revisit the logging changes made in PR #328. I'm changing some ERRORS and INFO to warnings in that issue to quiet some of the message spew due to the lack fo message filters. If we can enable message filters, these should quiet down anyway, and those log messages can go from DEBUG back to ERROR / INFO.

mkhansen-intel, 2018-11-17:
Also need to revert the change in #340 when message filters are working

gaoethan, 2018-11-23:
My porting  (https://github.com/ros2/geometry2/pull/81) to tf2 message filters has already been merged into geometry2, you can try to use it now and please file issue while you find.

crdelsey, 2018-11-28:
Probably #373 should be backed out when message filter functionality is re-enabled.