ros-navigation/navigation2 #191: Add Support for GridMap in Map_Server

bpwilcox, 2018-10-18:
To at the least demonstrate our extensible framework in the nav2_map_server, we should support [grid_map](https://github.com/ANYbotics/grid_map). An implementation of grid_map type maps would implement the interface in the BaseMapServer abstract class. Future consideration will be given to how clients of the map_server can potentially use a different map type. This is likely a discussion for the world model concept. #18

crdelsey, 2019-07-01:
This might come back when we deal with outdoor navigation or ramps

closed 2019-07-01 by crdelsey

SteveMacenski, 2019-07-10:
Side note: completely nuking costmap_2d is in my queue in the next year to do and replace with grid_maps. Why maintain a 10 year old technology when we could let the nice folks at Zurich do it, and they have alot of nice stuff like iterators that are objectively superior

crdelsey, 2019-07-11:
@SteveMacenski 
> completely nuking costmap_2d is in my queue in the next year to do and replace with grid_maps

Just FYI. @DLu has also completely redone costmap in his repo. That's another alternative to consider.

SteveMacenski, 2019-07-11:
Their costmap version is kind of a mid-way between ETH-Zurich's and the existing costmap_2d, I'll have to chat with him to see if there's something additional gained but tentatively my understanding is that grid_maps is more powerful given that we know its being used to represent both costs, gradients and traversibility maps (which was one of the design goals of navigation2). I think his work could be transitioned over to it but there's just an overwhelming amount of work done for us in grid_maps

justin-maidbot, 2022-04-14:
Based on the conversation in this thread + https://github.com/ros-planning/navigation2/issues/565 + https://github.com/ros-planning/navigation2/issues/1173, as well as going through some of the source code, it looks like nav2 map server and planners have costmaps hardcoded into the framework for now? Since these issues are a couple years old at this point, I was just curious if there's been any further work on integrating ANYbotics's GridMap into nav2 (https://github.com/ANYbotics/grid_map)?

SteveMacenski, 2022-04-14:
None currently, its a future project but hasn't been executed on yet