ros-navigation/navigation2 #193: Port voxel layer in nav2_costmap_2d

bpwilcox, 2018-10-18:
The voxel layer plugin remains to be ported to ROS2 in the costmap_2d package. The package currently compiles without including this layer. This layer derives from the obstacle layer (which as of this issue is not yet ported), so those dependencies would need to be resolved first.

prasenjitdan, 2018-10-24:
Hi @bpwilcox, In addition to our communication on https://discourse.ros.org/t/voxel-grid-porting-to-ros2/6581, I'm posting here as well. We are happy to help you in porting voxel-grid, so please let me know, if you will be willing to obtain our assistance, and if so, what precisely you would be willing us to take up.

crdelsey, 2018-10-24:
There's a voxel_grid package that is part of the ROS 1 navstack. I ported that over to ROS 2 before we started working in the public github repo. I'm assuming it is a dependency to getting the costmap layer working.

If you do in fact need it, let me know and I'll bring it over to this repo. It probably needs some updating to follow our current conventions.

SteveMacenski, 2018-10-24:
You'll need 3 things

- my PR (giving the observiation buffer) #231 
- Obstacle layer (which the voxel layer derives from)
- and Voxel_Grid (seems you have done)

bpwilcox, 2018-10-31:
@prasenjitdan If you're able to coordinate with @SteveMacenski who is working on the obstacle layer, then we'd be glad to accept your contribution on porting the voxel layer plugin.

SteveMacenski, 2018-10-31:
I posted on the obstacle layer ticket (https://github.com/ros-planning/navigation2/issues/192) the branch and status. I'm more than happy to let @prasenjitdan finish the port - it consists of dealing with the parameter while loop semantics and searchign for parameters. I don't know that I'll have the bandwidth to finish that one up anytime soon

bpwilcox, 2018-11-10:
@prasenjitdan Since I am going to commit to finishing the obstacle layer port myself, I could continue to do the voxel layer as well, unless you can commit to the port with a timeline in mind for us.

bpwilcox, 2018-11-14:
Resolved with #322

closed 2018-11-14 by bpwilcox