ros-navigation/navigation2 #199: Mission Execution defined non-serialized XML message

SteveMacenski, 2018-10-18:
I understand the purpose and value of the behavior tree, but I think its important to provide a real message type that can be used by code to populate the message not involving XML. The Path message in ROS1 did such a thing by just having an array of poses. If its important to include the named tag, it should be possible to make a type that contains a string field for "name" or "task" to give that custom XML-task-ed knowledge about what the metadata (geometry_msgs/PoseStamped) means to do with it. 

This might then become a living message type where as applications expand we add fields like the [rviz markers](http://docs.ros.org/melodic/api/visualization_msgs/html/msg/Marker.html) message. It's not ideal, but having a serialized string of XML isn't scalable or really usable for many applications. The ROS CLI tools for populating a message to test a result is important and with this interface, it would not be possible without writing nested XML by hand.  

In reference to this message and XML example, as well as comment: https://github.com/ros-planning/navigation2/blob/29a505ee1dc9cf14b116a3ea9685e80bf8f74921/nav2_mission_executor/README.md#overview. 
https://github.com/ros-planning/navigation2/pull/183#discussion_r226116336

mkhansen-intel, 2018-10-23:
Steve - I think this really should be something like "Define mission plan message type to be something other than string", because that's what I think you're asking for here, correct? 

The path message type isn't changing. The only place that a string is being used is as input to the mission executor.

I'm not sure what the BT XML schema is but we could potentially encode those fields into the message type directly. We'll take a look at that and update this ticket with a proposal.

SteveMacenski, 2018-10-23:
But there is though a subset of tasks to be completed, and each task probably has a bunch of the same metadata (location, tolerance) and then a string field for a name. This lets you analog the tag to a string field but still use defined messages you can introspect. Then those strings can contain any number of tasks of interest like docking, navigating, etc.

Alternatively, I would recommend just making a `UINT TASK=0` definitions in the message of common task types and continue to add more as useful ones pop up to use. Then people with custom messages can make their numbers high while the lower values are operations of general use

SteveMacenski, 2018-11-05:
From the WG meeting last week, we decided to keep it as a string

closed 2018-11-05 by SteveMacenski