ros-navigation/navigation2 #200: Time monitoring of time-sensitive tasks

mhpanah, 2018-10-18:
Time sensitive tasks such as global or local planner that are being run periodically may be required to be completed within some time window.  Therefore, it is important to implement a mechanism to monitor these tasks to ensure the time requirement is met.  If the time it takes to complete a task is longer than what is expected, the time monitor module may signal the mission executor and at that point the mission executor can decide what action to take.

mhpanah, 2019-07-23:
@mjeronimo It seems like this issue is very closely related to your performance work.  Should we close this issue for now?

SteveMacenski, 2020-03-29:
DWB reports on logging it misses rate: https://github.com/ros-planning/navigation2/blob/e15d8f228d8c9814506099e22aacce5ce6d7400a/nav2_controller/src/nav2_controller.cpp#L271

Maybe add something in the navigation action message feedback about the amount of time it took and if it was over? 

Navfn does not report anything. I'm working on a planner though that would give that ability. Navfn could be modified to do that but it would be somewhat challenging to limit the time vs just report it. If reporting is fine, then we can do that. Also then add to the action feedback about time spent. 

Behavior tree rate should be tracked somehow too in case its being stalled out. 

What else might want to report or limit time? I think those are the big tasks. Recoveries seems short enough its not worth it (thoughts?).

SteveMacenski, 2020-05-01:
We should also add the amount of time the planner took to compute to the planner server result

gokul-gokz, 2020-05-05:
I will work on this issue.

SteveMacenski, 2020-05-14:
Hi @gokul-gokz any updates?

SteveMacenski, 2020-07-01:
Added time information to the actions and the planner server now will give warnings when too slow

closed 2020-07-01 by SteveMacenski