ros-navigation/navigation2 #201: Assess, simplify and document the Navigation Stack parameters

crdelsey, 2018-10-18:
There are many parameters used in the current code base. Many of them are obsolete now.

We need to determine which parameters can be removed outright, which could be autoconfigured instead, and which are still needed.

Any remaining parameters needed to be documented, provided with reasonable defaults, and/or guides on how to configure them.

orduno, 2018-10-22:
Here are the parameters a user might want to tweak during run time for the Global Planner:

visualize_potential (bool)  // Whether we publish the potential array for visualization
allow_unknown (bool)  // Whether or not the planner should be allowed to plan through unknown space
default_tolerance (double)  // Amount the planner can relax the space constraint

bpwilcox, 2018-10-22:
Below are the parameters accessed inside the nav2_costmap_2d.

Costmap2DROS parameters:
- global_frame
- robot_base_frame
- rolling_window
- track_unknown_space
- always_send_full_costmap
- plugin_names
- plugin_types
- footprint_topic
- published_footprint_topic

Dynamic parameters for Costmap2DROS:
- _transform_tolerance_
- _update_frequency_
- _publish_frequency_
- _width_
- _height_
- _resolution_
- _origin_x_
- _origin_y_
- _footprint_
- _robot_radius_
- _footprint_padding_

Dynamic parameters for static layer plugin:
- _enabled_

Dynamic parameters for inflation layer plugin:
-  _enabled_
- _cost_scaling_factor_
- _inflation_radius_
- _inflate_unknown_

Dynamic parameters for obstacle layer plugin:
- _enabled_
- _footprint_clearing_enabled_
- _max_obstacle_height_
- _combination_method_

Dynamic parameters for voxel layer plugin:
- _enabled_
- _footprint_clearing_enabled_
- _max_obstacle_height_
- _origin_z_
- _z_resolution_
- _z_vowels_
- _unknown_threshold_
- _mark_threshold_
- _combination_method_

mhpanah, 2018-10-23:
Below are the parameters for AMCL.  These parameters are specifically for AMCL.

- use_map_topic_
- first_map_only_
- gui_publish_rate
- save_pose_rate
- laser_min_range
- laser_max_range
- max_beams
- min_particles
- max_particles
- pf_err
- pf_z
- alpha1, alpha2, alpha3, alpha4, alpha5
- do_beamskip
- beam_skip_distance
- beam_skip_threshold
- beam_skip_error_threshold
- z_hit
- z_short
- z_max
- z_rand
- sigma_hit
- lambda_short
- laser_likelihood_max_dist
- laser_model_type
- tmp_model_type
- update_min_d
- update_min_a
- odom_frame_id
- base_frame_id
- global_frame_id
- resample_interval
- transform_tolerancerecovery_alpha_slow
- recovery_alpha_fast
- tf_broadcast
- bag_scan_period

mjeronimo, 2018-10-24:
Mission Executor
- _No parameters_

Simple Navigator
- _No parameters_

BT Navigator
- _No parameters_

Tasks
- _No parameters_

AStar Planner
- Currently just a shell, _no parameters_

Robot
- _No parameters_

Map Server (currently, the map server takes the map type and the YAML filename on the command line. I'm working on a change where these would be parameters)
- _map_type_ - "occupancy", etc.
- _yaml_filename_ - the full path to the YAML file for the map

crdelsey, 2018-10-24:
# DWB Parameters
## Dynamic Parameters
* min_vel_x
* max_vel_x
* min_vel_y
* max_vel_y
* max_vel_theta
* acc_lim_x
* acc_lim_y
* acc_lim_theta
* decel_lim_x
* decel_lim_y
* decel_lim_theta
* min_speed_xy
* max_speed_xy
* min_speed_theta
* max_rot_vel (deprecated. Replaced by max_vel_theta)
* min_trans_vel (deprecated. Replaced by min_speed_xy)
* max_trans_vel (deprecated. Replaced by max_speed_xy)
* min_rot_vel (deprecated. Replaced by min_speed_theta)

## Other Parameters
### Related to which plugins to load
* critics
* prune_plan
* prune_distance
* trajectory_generator_name
* goal_checker_name
* default_critic_namespaces
* <plugin name>/class

### Related to which debug/visualization data to publish
* publish_evaluation
* publish_global_plan
* publish_local_plan
* publish_trajectories
* publish_cost_grid_pc

### base_obstacle critic configuration
* sum_scores

### map_grid critic configuration
* aggregation_type

### oscillation critic configuration
* x_only_threshold

### prefer_forward critic configuration
* penalty
* strafe_x
* strafe_theta
* theta_scale

### Various critics
* scale
* PathAlign/scale
* GoalAlign/scale
* PathDist/scale
* GoalDist/scale
* ObstacleFootprint/scale
* ObstacleFootprint/max_scaling_factor
* ObstacleFootprint/scaling_speed
* path_distance_bias (Deprecated. Replaced by PathAlign/scale and PathDist/scale)
* goal_distance_bias (Deprecated. Replaced by GoalAlign/scale and GoalDist/scale)
* occdist_scale (Deprecated. Replaced by ObstacleFootprint/scale)
* max_scaling_factor (Deprecated. Replaced by ObstacleFootprint/max_scaling_factor)
* scaling_speed (Deprecated. Replaced by ObstacleFootprint/scaling_speed)

### trajectory generator backward compatibility
* use_dwa

### limited_acceleration trajectory generator configuration
* sim_period

### simple goal checker configuration
* xy_goal_tolerance
* yaw_goal_tolerance

### standard trajectory generator configuration
* sim_time
* discretize_by_time
* controller_frequency
* time_granularity
* sim_granulairty (deprecated. Replacement is time_granularity)
* linear_granularity
* sim_granularity (deprecated. Replace by linear_granularity)
* angular_granularity
* angular_sim_granularity (deprecated. Replaced by angular_granularity)

### stopped goal checker configuration
* rot_stopped_velocity
* trans_stopped_velocity

### xy_theta_iterator configuration
* vx_samples
* vy_samples
* vtheta_samples
* vth_samples (deprecated. Replaced by vtheta_samples

### general
* odom_topic

### parameters with special handling
#### goal_align and path align critic configuration
* forward_point_distance

#### oscillation critic configuration
* oscillation_reset_dist
* oscillation_reset_angle
* oscillation_reset_time

#### rotate to goal critic
* xy_goal_tolerance

closed 2018-12-10 by mkhansen-intel