ros-navigation/navigation2 #202: Add support for additional local planner algorithms

crdelsey, 2018-10-18:
To ensure the navigation stack interfaces are complete and general enough, additional local planners that are substantially different from DWA should be ported to ROS2 and the new architecture.

 Possibilities for consideration include:
* [D* Lite](https://en.wikipedia.org/wiki/D*#D*_Lite)
* [5-D Planning with A*](http://ais.informatik.uni-freiburg.de/teaching/ss11/robotics/slides/18-robot-motion-planning.pdf)
* [Timed Elastic Band](http://wiki.ros.org/teb_local_planner)
* Convolutional Neural Networks
* [Fuzzy Logic](https://pdfs.semanticscholar.org/dd2f/4a3e9edc161fa327d189ebbe40e24ea64203.pdf) and [alternate paper](https://ieeexplore.ieee.org/document/4276413)
* [RRT](https://en.wikipedia.org/wiki/Rapidly-exploring_random_tree)

crdelsey, 2018-10-18:
Good call on the carrot planner. Even if it doesn't push the envelope on the interfaces, it would make a good example planner, and might be actually useful in some simple use cases.

SteveMacenski, 2018-10-18:
That's generally the conventional "first step" when hacking any robot (mobile, arm, etc) to just make it track some arbitrary carrot pointer that can be changed on the fly. I think that's a generally good choice as an example, but really only useful to diff drive small robots (and even then, depending on feature set, only good for R&D).

I can revisit my code for it and see if I can generalize it enough for this repo

crdelsey, 2018-10-18:
That would be great. Thanks!

crdelsey, 2018-10-18:
Removed RRT and D*Lite.

crdelsey, 2019-10-01:
Closing since this isn't addressing a specific need. Work on TEB is covered by other issues.

closed 2019-10-01 by crdelsey