ros-navigation/navigation2 #216: nav2_amcl test_localization is failing

crdelsey, 2018-10-19:
Here's the ctest -V output
```
8: Test timeout computed to be: 20
8: -- run_test.py: extra environment variables:
8:  - TEST_LAUNCH_DIR=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_amcl/test/test_launch_files
8:  - TEST_EXECUTABLE=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_amcl/test/integration/test_localization_node
8: -- run_test.py: invoking following command in '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_amcl/test/integration':
8:  - /home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_amcl/test/integration/test_localization_launch.py
8: [INFO] [launch]: process[amcl-1]: started with pid [6894]
8: [INFO] [launch]: process[map_server-2]: started with pid [6895]
8: [INFO] [launch]: process[test_localization_node-3]: started with pid [6896]
8: [ERROR] [launch]: process[amcl-1] process has died [pid 6894, exit code -6, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/install/nav2_amcl/lib/nav2_amcl/amcl'].
8/8 Test #8: test_localization ................***Timeout  20.01 sec

```

mhpanah, 2018-10-19:
nav2_amcl_test_localization is not an automated test.  To run this test, Gazebo and ros1_bridge needs to be running.

mkhansen-intel, 2018-11-12:
We now have Gazebo running in ROS2 so we should be able to enable this.

mkhansen-intel, 2018-12-03:
Fixed by #388

closed 2018-12-03 by mkhansen-intel