ros-navigation/navigation2 #219: Sort out copyright in eig3.* files

crdelsey, 2018-10-19:
nav2_util/include/nav2_util/pf/eig3.h and nav2_util/src/pf/eig3.c don't have clear copyright shown.

This needs to be address, and when it is, the cpplint checker needs to be re-enabled in the nav2_utils package.

mkhansen-intel, 2018-11-16:
@mikeferguson - do you know where these eig3.* files originated from? I see them in the navigation repo and they've been there awhile (2013 at least). Is it safe to bypass the copyright lint checker on these? They have no copyright or license info.

mikeferguson, 2018-11-16:
It looks like they came from https://math.nist.gov/javanumerics/jama/ based on the comments on eig3.h -- I'd suggest maybe bringing the actual copyright info over from that page?

mkhansen-intel, 2018-11-16:
This is the copyright on that page:
Copyright Notice This software is a cooperative product of The MathWorks and the National Institute of Standards and Technology (NIST) which has been released to the public domain. Neither The MathWorks nor NIST assumes any responsibility whatsoever for its use by other parties, and makes no guarantees, expressed or implied, about its quality, reliability, or any other characteristic.

@crdelsey - do you know if this will pass the linter? It seems like it's the right thing to do to include it.

I don't see any type of license info on that website.

crdelsey, 2018-11-16:
The C rewrite is not the Java version, so could have a different copyright. Unfortunately, this file predates the oldest commit in the git history, so I can't tell who wrote it.

Based on the comment that @mikeferguson mentioned, it seems that whoever wrote it intended to release this under the same license, ie public domain, so I think that is the best we can do until the original author speaks up and gives us their real intent.

This won't pass the linter. We'll have to disable that check.

PR #403 (2018-12-05): Make nav2_utils pass colcon test.
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | Resolves #219. Partially addresses #334 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

* Fixes uncrustify issues in nav2_utils
* Verifies copyright test is disabled.
* Enables nav2_utils testing in travis.
closed 2018-12-05 by mkhansen-intel