ros-navigation/navigation2 #236: Refactoring AMCL

mhpanah, 2018-10-24:
@SteveMacenski and @mikeferguson I am planning to make few changes to AMCL and would like to get your feedback before committing.  Please look at the items below and let me know your thoughts.

**AMCL's sensor measurements base class**

1. Since the Probabilistic Robotics (PR) textbook is considering the odometry as if it was a control signal and it is treating the odometry measurements as controls, I am proposing to bring out the AMCLodom from the Sensor base class. 

2. Once, we bring out the odom from sensor base class, since there are no other measurement classes, I propose to remove the AMCL Sensor base class.  In future if we create a class for other sensors such as Sonar or etc. we can then add a base class.

**AMCLOdom**
I am proposing to rename the AMCLOdom to robot_motion_model.  The robot motion model can have Velocity Motion Model and Odometry Motion Model (ODM) in which the ODM can have differential and omni-directional types for motion update methods. This way the AMCL code would be more aligned with PR and we can later integrate it into Robot class.

**amcl_node.cpp**
1. I usually define Xt as `current_pose` and Xt_m1 as `previous_pose`.  I would like to change the `old_pose` to `previous_pose` or `prev_pose`.

2. `pf_vector_t delta`.  In the `updateAction` method, I’m proposing to pass `current_pose` and `previous_pose` rather than `delta`.  This way the code would be more readable as it is more similar to PR and the AMCL core code would be a bit cleaner.

3. getOdom method

```
 if (!getOdomPose(latest_odom_pose_, pose.v[0], pose.v[1], pose.v[2],
    laser_scan->header.stamp, base_frame_id_))
  {
    RCLCPP_ERROR(get_logger(), "Couldn't determine robot's pose associated with laser scan");
    return;
  }
```

If I'm not mistaken, here, `latest_odom_pose_` and `pose.v[0], pose.v[1], pose.v[2]` are the same thing. Please correct me if that’s not the case. Therefore, there is no point to have both of them in the getOdomPose method.   Also, `latest_odom_pose_` should be changed to `current_pose`, if that's what it is.  
**Question:** Does the `latest_odom_pose_` coming from robots IMU or is it the last pose that was estimated from AMCL?

**pf_vector.cpp**
1. How about utilizing Eigen library instead?

SteveMacenski, 2018-10-25:
> AMCL's sensor measurements base class

I don't fully understand the motive. The Sensor and SensorData classes are interfaces for AMCL to interact with models to allow someone to swap out either the motion model (odom) or sensor model (laser) for another model for each. I agree perhaps the name Sensor may not be appropriate, but having some kind of interface to fill out your motion/sensor models is needed for someone to come by and implement their own with a structured API.

It could be imaginable for there to be multiple sensor models wanted to weigh particles other than a laser (potentially adding in visual features, etc) and for motion models based on different base types or mathematical formulation.

 > AMCLOdom

I think that rename is good. I don't understand your distinction between Odometry and Velocity models, I would consider velocity a component of odometry.

> amcl_node.cpp

No arguments on first 2. On point 3, that function is taking in the latest_odom_pose_ and then filling out the pose object. It looks like it's looking up an identity transform but then I'm also not seeing anywhere latest_odom_pose_ is actually being updated to make that work. Overall a little confusing.

mikeferguson, 2018-10-25:
> > AMCLOdom
> 
> I think that rename is good. I don't understand your distinction between Odometry and Velocity models, I would consider velocity a component of odometry.
> 

Probabilistic Robotics book shows two different models for odometry -- basically, do you integrate the position changes OR do you integrate the velocity at each step. Depending on the quality of your estimates, one model might be better than the other.

mikeferguson, 2018-10-25:
> **AMCL's sensor measurements base class**
>
>    1. Since the Probabilistic Robotics (PR) textbook is considering the odometry as if it was a control signal and it is treating the odometry measurements [...]

I'd suggest keeping the base classes for sensor (also where is the laser? is that not a sensor?). There are several forks of AMCL that (hackily) implement other sensors (wifi, sonars, etc). Having a base class for sensors makes it very clear how to add other sensors, maybe that eventually gets those wifi/etc versions merged here (it was always too difficult in the existing ROS1 AMCL implementation, since it touched so much code).

> **AMCLOdom**
> I am proposing to rename the AMCLOdom to robot_motion_model. The robot motion model can have Velocity Motion Model and Odometry Motion Model (ODM) in which the ODM can have differential [...]

LGTM

>    1. I usually define Xt as `current_pose` and Xt_m1 as `previous_pose`.  I would like to change the `old_pose` to `previous_pose` or `prev_pose`.

LGTM
 
>    2. `pf_vector_t delta`.  In the `updateAction` method, I’m proposing to pass `current_pose` and `previous_pose` rather than `delta`.  This way the code would be more readable as it is more similar to PR and the AMCL core code would be a bit cleaner.

LGTM

> 3. getOdom method
>
> ```
>  if (!getOdomPose(latest_odom_pose_, pose.v[0], pose.v[1], pose.v[2],
>     laser_scan->header.stamp, base_frame_id_))
>   {
>     RCLCPP_ERROR(get_logger(), "Couldn't determine [...]

First: the latest_odom_pose_ is actually the base_link->odom transform, found through TF2 -- so that comes from whatever is creating your odometry (might be wheel encoders, might be IMU, might be a fusion of both, for instance using robot_localization package).

AFAIK, the only place the latest_odom_pose_ is used, is to save it to the param server on shutdown, so that you don't lose the robot position.

And yes, the data is the same, although pose.v[x] is a subset of the actual data. It'd probably be more generic to have a second function that converts from geometry_msgs/Pose to the particle type.

> **pf_vector.cpp**
> 
>     1. How about utilizing Eigen library instead?

I think replacing any custom matrix code with Eigen makes sense -- remember, this code is ~20 years old, so I think it predates Eigen by a bunch...

mhpanah, 2018-10-25:
> Probabilistic Robotics book shows two different models for odometry -- basically, do you integrate the position changes OR do you integrate the velocity at each step. Depending on the quality of [...]

Yes PR book has two different models. Please look at slides 24 and 30 for details. 
http://ais.informatik.uni-freiburg.de/teaching/ss11/robotics/slides/06-motion-models.pdf

mhpanah, 2018-10-25:
Thanks for feedback and clarifications!  I will address them on my next pull request. 

- As you guys suggested keeping the sensor base class make sense. However, to make it less confusing I can create two base classes. One for sensors (laser, sonar, wifi, etc.) and one for robot motion model update (Omni, Differential, etc).  

- For now I will keep the pf_vector class, but replacing that with eigen should be something to keep in mind.

mhpanah, 2019-01-09:
Have already addressed most of the items in this issue. We may reopen this issue if we need to.

closed 2019-01-09 by mhpanah