ros-navigation/navigation2 #274: [Question] What does the intel ros2_message_filters do that tf2_ros versions dont?

SteveMacenski, 2018-10-31:
I see we're building it in the ros2_dependencies file. Is anyone dependent on this package and how does it differ from tools that exist already of similar name?
 
If we actually need it, then someone at Intel needs to commit to adding the rosdep keys for it and bloom releasing it so we can build the anvigation2 stack in CI.

mkhansen-intel, 2018-11-12:
https://github.com/ros2/message_filters/pull/13

mkhansen-intel, 2018-11-12:
@gaoethan - Is there any reason we can't replace intel/message_filters with ros2/message_filters when the PR above (#13) is merged?

gaoethan, 2018-11-13:
Actually, the current` ros2/message_filters` is derived from our `intel/message_filters`, we have already formally transferred our porting to ros2 community and maintain it as `ros2/message_filters`, so please use the `ros2/message_filters` now, thanks !

PR #402 (2018-12-04): Remove the old message filters dependency.
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | Resolves #274 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | TB3 in gazebo |

---

## Description of contribution in a few bullet points

* Removes message filters from the additional ROS 2 dependencies. It is part of base ROS 2 now.
closed 2018-12-04 by crdelsey