ros-navigation/navigation2 #285: Remove dependency on custom ros2/launch

crdelsey, 2018-11-01:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Summary
The nav2_bringup repo uses TimerActions in the launch file to slow down the process and bring everything up reliably. Unfortunately, TimerActions are not supported yet in the official ros2/launch repo, so we are using a custom version of Launch.

Before we can go into the crystal release and before the ROS build farm can build our stack, this dependency needs to be removed, and the launch files need to work with the official launch tool

mkhansen-intel, 2018-11-26:
@crdelsey - since we're not using TimerActions, I think we can do this now, correct?

crdelsey, 2018-11-26:
Yes. I'll do this now.

PR #361 (2018-11-26): Removing dependency on custom ros2 launch.
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | Resolves #285. Resolves #348. |
| Primary OS tested on | NA |
| Primary platform tested on | NA |

--- 

## Description of contribution in a few bullet points
* Removes the dependency on  the customized ROS 2 launch tool.
* Removes the obsolete launch file that still has the `TimerAction` actions
closed 2018-11-26 by mkhansen-intel