ros-navigation/navigation2 #295: nav2_map_server and nav2_world_model do not shutdown on Ctrl-C (SIGINT)

crdelsey, 2018-11-07:
## Bug report

**Required Info:**

#### Steps to reproduce issue

```bash
roslaunch nav2_bringup nav2_bringup_launch.py map:=/home/crdelsey/map.yaml
```
Then, when everything is up, hit Ctrl-C.
#### Expected behavior
```
[INFO] [launch]: process[world_model]: process has finished cleanly
[INFO] [launch]: process[map_server]: process has finished cleanly
```
#### Actual behavior
```
[ERROR] [launch]: process[world_model-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [launch]: process[map_server-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [launch]: sending signal 'SIGTERM' to process[world_model-2]
[INFO] [launch]: sending signal 'SIGTERM' to process[map_server-1]
[ERROR] [launch]: process[map_server-1] process has died [pid 14897, exit code -15, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/install/nav2_map_server/lib/nav2_map_server/map_server /home/crdelsey/map.yaml occupancy'].
[ERROR] [launch]: process[world_model-2] process has died [pid 14898, exit code -15, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/install/nav2_world_model/lib/nav2_world_model/world_model'].

```
#### Additional information

mkhansen-intel, 2018-12-10:
Fixed for map_server, not yet for world_model

crdelsey, 2019-01-22:
This seems to be all fixed now. Closing.

closed 2019-01-22 by crdelsey