ros-navigation/navigation2 #317: Make Travis fail on broken builds

Nickolaim, 2018-11-13:
Currently, even if a build is broken, Travis shows green status.  For example, job [https://travis-ci.org/ros-planning/navigation2/jobs/454313681](https://travis-ci.org/ros-planning/navigation2/jobs/454313681), raw log [https://api.travis-ci.org/v3/job/454313681/log.txt](https://api.travis-ci.org/v3/job/454313681/log.txt), toward the end of the log.

```
Summary: 26 packages finished [8min 6s]
  4 packages had stderr output: dwb_msgs nav2_msgs nav_2d_msgs navigation2
```
The proposal is to fail Travis build if `colcon build` in `navigation2` fails.

If you are fine with the proposal, I can work on this.

SteveMacenski, 2018-11-13:
I dont see any errors that should make that build fail - those are primarily just warnings

Nickolaim, 2018-11-15:
Interesting.  I will read more about the build system in ROS2. 

Thanks for the clarification.

closed 2018-11-15 by Nickolaim