ros-navigation/navigation2 #326: Port costmap_2d_markers and costmap_2d _cloud visualization tools

bpwilcox, 2018-11-15:
There are two costmap_2d executables ([costmap_2d_markers](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_markers.cpp) and [costmap_2d_cloud](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_cloud.cpp)) which have yet to be ported. They are not part of the critical navigation path, but  (to my knowledge) provides publishers for voxel grid visualization.

SteveMacenski, 2018-11-15:
Thoughts on just including them in the voxel layer itself? The cloud uses a `sensor_msgs::PointCloud` which has long been replaced by most folks with `sensor_msgs::PointCloud2`. Markers could be reused elsewhere though I dont know that anyone actually is

crdelsey, 2019-07-01:
This is done now.

closed 2019-07-01 by crdelsey