ros-navigation/navigation2 #331: map_server is continuously looping to publish map

mkhansen-intel, 2018-11-15:
https://github.com/ros-planning/navigation2/blob/c5db53b29c5ce9a37199b92dbc6dc8670cbfe5dc/nav2_map_server/src/map_representations/occ_grid_server.cpp#L323

There is a todo comment to fix this when everything is in ROS2, now that everything is in ROS2, can we remove this?

mkhansen-intel, 2018-11-15:
Linked to #328 since one of the changes in that PR is due to this. If this is removed, we can remove that change too.

SteveMacenski, 2018-11-16:
In particular, when latched topics are back in move the mapReceived to a `RCLCPP_INFO` and when the message filters are around, warnings and errors should be put back in place rather than DEBUGs. (just being verbose for my own posterity once those are around)

mkhansen-intel, 2019-09-18:
The "TODO" is no longer there and this can be closed. The republishing is still there but will be removed when #867 is fixed, so this is redundant.

SteveMacenski, 2019-09-18:
@mkhansen-intel are latched topics in ROS2 / being used in map server? I dont recall that change coming in

closed 2019-10-10 by SteveMacenski