ros-navigation/navigation2 #334: 20 packages failing colcon test need lint / uncrustify fixes

mkhansen-intel, 2018-11-16:
running colcon test on master today:
```
Summary: 26 packages finished [8.11s]
  20 packages had stderr output: costmap_queue dwb_controller dwb_core dwb_critics dwb_plugins nav2_amcl nav2_bt_navigator nav2_costmap_2d nav2_dynamic_params nav2_map_server nav2_mission_executor nav2_robot nav2_simple_navigator nav2_smart_planner nav2_system_tests nav2_tasks nav2_util nav2_voxel_grid nav2_world_model nav_2d_utils
  20 packages had test failures: costmap_queue dwb_controller dwb_core dwb_critics dwb_plugins nav2_amcl nav2_bt_navigator nav2_costmap_2d nav2_dynamic_params nav2_map_server nav2_mission_executor nav2_robot nav2_simple_navigator nav2_smart_planner nav2_system_tests nav2_tasks nav2_util nav2_voxel_grid nav2_world_model nav_2d_utils
```
To reproduce:
`colcon test`
To see log
`less ~/ros2_ws/navigation2_ws/log/latest_test/<package_name>/stdout_stderr.log`

We need to divide and conquer to get these fixed.

crdelsey, 2018-11-19:
Related to #220, #221, #215, and #216

Nickolaim, 2018-11-22:
Feel free to assign to me.  Not sure if this is going to be done in Nov, since there are a lot of fixes to be made.

Nickolaim, 2018-11-23:
I tried to debug this.  I started with `costmap_queue` package.  In my machine, it only has the problem with failing `uncrustify`.

Here is the example of how it is failing on my machine: 
```
/home/nick/nav2/ros2_ws/build/ament_uncrustify/ament_uncrustify/configuration/ament_code_style.cfg:136 Unknown symbol 'sp_after_ptr_block_caret'
/home/nick/nav2/ros2_ws/build/ament_uncrustify/ament_uncrustify/configuration/ament_code_style.cfg:253 Unknown symbol 'sp_cpp_before_struct_binding'
/home/nick/nav2/ros2_ws/build/ament_uncrustify/ament_uncrustify/configuration/ament_code_style.cfg:263 Unknown symbol 'sp_inside_square_oc_array'
...
```

The reason for the failure is that `uncrustify` is called from the wrong location, and in my case, the version that is called is 0.66, but these settings are only get added in 0.67.  The right version of `uncrustify` is in `./home/nick/nav2/ros2_ws/install/uncrustify_vendor/bin/uncrustify`

If I run the correct version of the tool, I get the nice output.

```
$ /home/nick/nav2/ros2_ws/install/ament_uncrustify/bin/ament_uncrustify --xunit-file /home/nick/nav2/navigation2_ws/build/costmap_queue/test_results/costmap_queue/uncrustify.xunit.xml
No code style divergence in file 'include/costmap_queue/costmap_queue.hpp'
No code style divergence in file 'include/costmap_queue/limited_costmap_queue.hpp'
No code style divergence in file 'include/costmap_queue/map_based_queue.hpp'
No code style divergence in file 'src/costmap_queue.cpp'
No code style divergence in file 'src/limited_costmap_queue.cpp'
No code style divergence in file 'test/mbq_test.cpp'
No code style divergence in file 'test/utest.cpp'
No errors
```

I want to get info on are you getting the same error.  If yes, this is an environmental/configuration problem.

mkhansen-intel, 2018-11-26:
@Nickolaim - First, thanks for taking a look at this. 

Second, I'm not sure how to help as I don't know what command you tried in the first (failing) case. If you could add that I could try it too. 

I did check which uncrustify I am using by default and it's this one:
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ which uncrustify
/home/mkhansen/ros2_dev/ros2_ws/install/uncrustify_vendor/bin/uncrustify
```
I know that the version of uncrustify changed, as did the location, between the bouncy release and now. If you installed and built ROS2 from the 'master' version of the ros2.repos file, you should be getting the latest one, if you installed and built from the 'bouncy' version that's probably what's causing this problem. 

Hope that helps. If you need more help, please post the command you ran when it failed.

Nickolaim, 2018-11-26:
@mkhansen-intel  - can you please share the stdout for `colcon test` for `costmap_queue` package?  It fails for me with errors in the config file.  If it fails the same way for you, it means the wrong version of uncrustify is running.

crdelsey, 2018-11-26:
@Nickolaim The `colcon test` command calls `ament_uncrustify`, not `uncrustify` so if you want to run the command manually, you need to have `install/setup.bash` sourced.

That said, `costmap_queue` passes for me locally as well.

Nickolaim, 2018-11-26:
@crdelsey `ament_uncrustify` calls `uncrustify`, see [https://github.com/ament/ament_lint/blob/17df968e6962ca4ea423e04b9d05123517eed116/ament_uncrustify/ament_uncrustify/main.py#L104](https://github.com/ament/ament_lint/blob/17df968e6962ca4ea423e04b9d05123517eed116/ament_uncrustify/ament_uncrustify/main.py#L104)

So depending on what version of `uncrustify` is being used, the test either passes for `costmap_queue` or fails.  If 0.68 version is used it passes; if 0.67 is used it is failing.  At least this is what I got in my investigation.  I believe the code in `ament_uncrustify` should check for the version of the tool before running, but that's a separate issue outside of this work, and I will follow up on it later.

The issue has been filed 11 days ago.  The last update to the code happened 21 days ago [https://github.com/ros-planning/navigation2/tree/master/nav2_dwb_controller/costmap_queue](https://github.com/ros-planning/navigation2/tree/master/nav2_dwb_controller/costmap_queue), so it is unlikely that the code has been fixed after the issue is filed.  I am trying to get confirmation from @mkhansen-intel that when he filed this issue he got the same errors as I mentioned above (in config file).

Nickolaim, 2018-11-27:
I think the way how `colcon test` was called in Travis CI needed improvement, and I submitted a PR for this.  After this change, `costmap_queue` is passing tests. I think it will be easier now to fix the remaining issues - remove the package from the skip list, inspect the failures, fix them and merge :)

crdelsey, 2019-01-22:
All packages are passing now.

closed 2019-01-22 by crdelsey