ros-navigation/navigation2 #339: AMCL fails to transform most laser scan data it receives

crdelsey, 2018-11-16:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2018-11-16:
@crdelsey If I recall correctly: you're not using a EKF/UKF/etc in order to get higher rate pose estimations - only basing it on AMCL's pose estimation that comes in at ~1hz or so depending on motion. This could cause what you're seeing.

crdelsey, 2018-11-16:
@SteveMacenski Specifically, the problem we see is we get to the transform call in getOdomPose and it fails most of the time. Even with the robot not moving. For @mkhansen-intel This transform succeeds once out of every ~30 times. For me is succeeds once in ~5000 time.

For me at least, it fails because `Lookup would require extrapolation into the future.`

SteveMacenski, 2018-11-16:
Yes, that sounds like what I was mentioning. Your transformation expires and if there's no republisher or EKF to keep giving you updates, then its going to fail between them.

AMCL is meant not as a state estimator but as a global landmark updater to recorrect for drift in your odometry (usually fused into a filter) that's publishing the dead-reckoned odometry into TF. If you're not publishing into TF then it will quickly go out of scope. While you're moving and you have data, if no one's telling TF, then you're marching forward in time, but the last piece of information you have about your robot's state was from AMCL, say, 2 seconds ago

crdelsey, 2018-11-16:
We've got gazebo publishing odom - base_footprint transforms every 10ms, so it's not that there are no updates once we start. If I understand you correctly, what Gazebo is doing should be the equivalent or better than the test you suggested.

SteveMacenski, 2018-11-17:
Yes, I didn't realize you were in simulation. Gazebo should be doing that if you have the plugin setup to publish TF. That can be verified by looking at the tf topic output but sounds like you have that all squared away.

Looking at that error (as you know) that leaves the other knee-jerk potential problem to be with ROS time not handling sim time correctly so that time its trying to look up a transform for is current. Can you publish the timestamps of the scan and request to check that everything is indeed registered/operating with sim time?

If that's not the issue, I'd have to look further into it.

crdelsey, 2018-11-17:
Yes. Everything looks like it is using sim_time, although it can sometimes be tricky to get everything switched over. 

My plan is spend Thanksgiving week digging into the guts of tf2 to figure out what's going on. I've stepped through the code a few times, but don't have a good understanding of the tf2 internal structures and algorithms to understand how it works with time.

crdelsey, 2018-11-27:
Fixed in #340

closed 2018-11-27 by crdelsey