ros-navigation/navigation2 #341: AMCL sometimes can't localize even when given initalpose

mkhansen-intel, 2018-11-17:
#### Steps to reproduce issue
start gazebo with the world file
```
gazebo --verbose -s libgazebo_ros_init.so <path to world file>
```
start RVIZ
```
ros2 run rviz2 rviz2
```
start the stack by running the nav2_bringup_launch
```
ros2 launch nav2_bringup nav2_bringup_launch.py map:=<path to map> use_sim_time:=True
```
set the use_sim_time to true for static_tf_publisher
```
ros2 param set /static_transform_publisher use_sim_time true
```
In another window, echo the amcl_pose topic:
```
ros2 topic echo /amcl_pose
```
In RVIZ, click on '2D pose estimate' and set it to the correct pose

#### Expected behavior
The amcl_pose will be published with a pose very near to what was sent by /initialpose
#### Actual behavior
Sometimes we get this:
header:
  stamp:
    sec: 8
    nanosec: 249000000
  frame_id: map
pose:
  pose:
    position:
      x: -2.0
      y: 4.336808689942018e-19
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0946799022804343e-13]

Note that the "y" position is crazy, as is the last covariance value. Not sure why this happens, but it happens more than half the time.

crdelsey, 2018-11-17:
The y value is e -19, so essentially 0. That might be OK. Just a floating point representation thing.

mkhansen-intel, 2018-11-18:
Yeah, I should have realized that. Something is still weird though because it doesn't seem to show up in RVIZ, I'll take a look again tomorrow (Monday).

mhpanah, 2018-11-21:
@mkhansen-intel I don't have this issue on my machine.  Do you still not see a correct initial pose?