ros-navigation/navigation2 #345: Remove "angles" from ros2_dependencies.repos (again)

mkhansen-intel, 2018-11-19:
Once the 'angles' repo dependency is in the Crystal rosdistro file, we need to remove it again from the ros2_dependencies.repos file.

This is basically a reminder to revert the change in #343.

mkhansen-intel, 2018-11-20:
I opened this issue https://github.com/ros2/rosdistro/issues/344 to request angles to be added to the Crystal distro file. When that is done, we can remove from our repos file.

SteveMacenski, 2018-11-26:
It wont be possible to release to bloom without all of the *.repos dependencies as rosdep keys and building in the build farm. There may be a way around it but would involve quite a bit of work. 

Angles doesnt need to be included in the ros2.repos file as long as there's rosdep keys for it - which it looks like Dirk just did

SteveMacenski, 2018-11-26:
@mkhansen-intel rosdep keys are back https://github.com/ros2/rosdistro/blob/ros2/crystal/distribution.yaml#L132

You may remove angles from the repos file

SteveMacenski, 2018-11-26:
https://github.com/ros-planning/navigation2/pull/360

crdelsey, 2019-03-07:
Due to the way we are building in CI - tracking the ROS2 nightly latest docker - we can't remove the navstack dependencies and rely on `rosdep` to bring everything in. `rosdep` doesn't know about the files already installed in the docker and, instead, overwrites them with the release crystal packages.

Instead, we want another build based on the `Dockerfile.crystal` that brings in all dependencies using rosdep to ensure all dependencies are released. I'm keeping this issue open as a reminder to do that.

SteveMacenski, 2019-05-11:
Given  then that we have other deps still in the ros2 file that don't appear to be going away anytime soon, we may just want to call this ticket closed.

@crdelsey can we just close it?

SteveMacenski, 2019-05-15:
Closing since the ros2 repo file isn't going away anytime soon. This was part of an effort to remove it, but I think after a couple of ROS2 release cycles in this project I don't think we'll be able to for a little while at least.

closed 2019-05-15 by SteveMacenski