ros-navigation/navigation2 #354: Navstack crashes at startup

mhpanah, 2018-11-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - Master crashes from ID: 1f44491
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
ros2 launch nav2_bringup nav2_bringup_launch.py map:=~/navigation2/map/map.yaml use_sim_time:=true

```

#### Expected behavior
Code runs without crashing
#### Actual behavior
It crashes
#### Additional information
```
[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): "package 'nav2_navfn_planner' not found, searching: ['/home/mohammad/OTC_workDir/ros2_ws/install/ament_uncrustify', '/home/mohammad/OTC_workDir/ros2_ws/install/topic_monitor', '/home/mohammad/OTC_workDir/ros2_ws/install/test_launch_ros', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclpy_minimal_subscriber', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclpy_minimal_service', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclpy_minimal_publisher', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclpy_minimal_client', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclpy_executors', '/home/mohammad/OTC_workDir/ros2_ws/install/demo_nodes_py', '/home/mohammad/OTC_workDir/ros2_ws/install/sros2', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2lifecycle', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2service', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2topic', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2srv', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2run', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2launch', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2pkg', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2param', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2node', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2multicast', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2msg', '/home/mohammad/OTC_workDir/ros2_ws/install/ros2cli', '/home/mohammad/OTC_workDir/ros2_ws/install/launch_ros', '/home/mohammad/OTC_workDir/ros2_ws/install/launch_testing', '/home/mohammad/OTC_workDir/ros2_ws/install/launch', '/home/mohammad/OTC_workDir/ros2_ws/install/osrf_pycommon', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_xmllint', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_pyflakes', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_pep8', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_pep257', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_pclint', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_package', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_lint_cmake', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_index_python', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_flake8', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_cpplint', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_cppcheck', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_copyright', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_clang_format', '/home/mohammad/OTC_workDir/navigation2/install/nav2_bringup', '/home/mohammad/OTC_workDir/navigation2/install/navigation2', '/home/mohammad/OTC_workDir/navigation2/install/nav2_dwb_controller', '/home/mohammad/OTC_workDir/navigation2/install/dwb_plugins', '/home/mohammad/OTC_workDir/navigation2/install/dwb_critics', '/home/mohammad/OTC_workDir/navigation2/install/dwb_core', '/home/mohammad/OTC_workDir/navigation2/install/nav_2d_utils', '/home/mohammad/OTC_workDir/navigation2/install/dwb_msgs', '/home/mohammad/OTC_workDir/navigation2/install/nav_2d_msgs', '/home/mohammad/OTC_workDir/navigation2/install/nav2_world_model', '/home/mohammad/OTC_workDir/navigation2/install/nav2_voxel_grid', '/home/mohammad/OTC_workDir/navigation2/install/nav2_system_tests', '/home/mohammad/OTC_workDir/navigation2/install/costmap_queue', '/home/mohammad/OTC_workDir/navigation2/install/nav2_costmap_2d', '/home/mohammad/OTC_workDir/navigation2/install/nav2_util', '/home/mohammad/OTC_workDir/navigation2/install/nav2_simple_navigator', '/home/mohammad/OTC_workDir/navigation2/install/nav2_mission_executor', '/home/mohammad/OTC_workDir/navigation2/install/nav2_bt_navigator', '/home/mohammad/OTC_workDir/navigation2/install/nav2_tasks', '/home/mohammad/OTC_workDir/navigation2/install/nav2_robot', '/home/mohammad/OTC_workDir/navigation2/install/dwb_controller', '/home/mohammad/OTC_workDir/navigation2/install/nav2_msgs', '/home/mohammad/OTC_workDir/navigation2/install/nav2_map_server', '/home/mohammad/OTC_workDir/navigation2/install/nav2_dynamic_params', '/home/mohammad/OTC_workDir/navigation2/install/nav2_amcl', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/xmlrpcpp', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/message_filters', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/gazebo_ros_pkgs', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/gazebo_plugins', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/gazebo_ros', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/gazebo_msgs', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/gazebo_dev', '/home/mohammad/OTC_workDir/navstack_dependencies_ws/install/angles', '/home/mohammad/OTC_workDir/ros2_ws/install/rviz_visual_testing_framework', '/home/mohammad/OTC_workDir/ros2_ws/install/rviz2', '/home/mohammad/OTC_workDir/ros2_ws/install/rviz_default_plugins', '/home/mohammad/OTC_workDir/ros2_ws/install/rviz_common', '/home/mohammad/OTC_workDir/ros2_ws/install/yaml_cpp_vendor', '/home/mohammad/OTC_workDir/ros2_ws/install/ros1_bridge', '/home/mohammad/OTC_workDir/ros2_ws/install/common_interfaces', '/home/mohammad/OTC_workDir/ros2_ws/install/visualization_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/dummy_robot_bringup', '/home/mohammad/OTC_workDir/ros2_ws/install/robot_state_publisher', '/home/mohammad/OTC_workDir/ros2_ws/install/kdl_parser', '/home/mohammad/OTC_workDir/ros2_ws/install/urdf', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_lint_common', '/home/mohammad/OTC_workDir/ros2_ws/install/ament_cmake_uncrustify', '/home/mohammad/OTC_workDir/ros2_ws/install/uncrustify_vendor', '/home/mohammad/OTC_workDir/ros2_ws/install/trajectory_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/pendulum_control', '/home/mohammad/OTC_workDir/ros2_ws/install/tlsf_cpp', '/home/mohammad/OTC_workDir/ros2_ws/install/tlsf', '/home/mohammad/OTC_workDir/ros2_ws/install/test_pluginlib_fixture', '/home/mohammad/OTC_workDir/ros2_ws/install/pluginlib', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2_sensor_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2_geometry_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2_eigen', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2_ros', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/laser_geometry', '/home/mohammad/OTC_workDir/ros2_ws/install/tf2', '/home/mohammad/OTC_workDir/ros2_ws/install/test_security', '/home/mohammad/OTC_workDir/ros2_ws/install/test_rclcpp', '/home/mohammad/OTC_workDir/ros2_ws/install/test_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/test_communication', '/home/mohammad/OTC_workDir/ros2_ws/install/test_cli_remapping', '/home/mohammad/OTC_workDir/ros2_ws/install/test_cli', '/home/mohammad/OTC_workDir/ros2_ws/install/stereo_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/std_srvs', '/home/mohammad/OTC_workDir/ros2_ws/install/shape_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/map_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/intra_process_demo', '/home/mohammad/OTC_workDir/ros2_ws/install/image_tools', '/home/mohammad/OTC_workDir/ros2_ws/install/dummy_sensors', '/home/mohammad/OTC_workDir/ros2_ws/install/sensor_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/dummy_map_server', '/home/mohammad/OTC_workDir/ros2_ws/install/nav_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/logging_demo', '/home/mohammad/OTC_workDir/ros2_ws/install/lifecycle', '/home/mohammad/OTC_workDir/ros2_ws/install/diagnostic_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/geometry_msgs', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclcpp_minimal_subscriber', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclcpp_minimal_publisher', '/home/mohammad/OTC_workDir/ros2_ws/install/examples_rclcpp_minimal_composition', '/home/mohammad/OTC_workDir/ros2_ws/install/demo_nodes_cpp_native', '/home/mohammad/OTC_workDir/ros2_ws/install/demo_nodes_cpp', '/home/mohammad/OTC_workDir [... cut ...]
[INFO] [launch]: process[amcl-5]: process has finished cleanly
[ERROR] [launch]: process[map_server-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [launch]: sending signal 'SIGTERM' to process[map_server-3]
[ERROR] [launch]: process[map_server-3] process has died [pid 16799, exit code -15, cmd '/home/mohammad/OTC_workDir/navigation2/install/nav2_map_server/lib/nav2_map_server/map_server /home/mohammad/OTC_workDir/navigation2/map/map.yaml occupancy'].
```

----

mkhansen-intel, 2018-11-21:
Did you rebuild and re-run install/setup.bash?

When I did that, this problem went away for me.

abylikhsanov, 2018-11-23:
What is the build status? I cannot compile due to the error in nav2_dynamic_params. 
```
navigation2/nav2_dynamic_params/include/nav2_dynamic_params/dynamic_params_client.hpp:240:29: error: ‘using element_type = class rclcpp::Node {aka class rclcpp::Node}’ has no member named ‘get_node_names’; did you mean ‘get_name’?
     auto node_list = node_->get_node_names();
                             ^~~~~~~~~~~~~~
                             get_name
/home/ikhsanov/ws/navigation2_ws/src/navigation2/nav2_dynamic_params/include/nav2_dynamic_params/dynamic_params_client.hpp:243:35: error: unable to deduce ‘auto&&’ from ‘node_list’
     for (const auto & node_name : node_list) {
```

mkhansen-intel, 2018-11-26:
@abylikhsanov - sorry I haven't seen this issue, if you're still seeing it, please file a new issue with your environment info and steps to reproduce and we'll take a look. Our build is passing.