ros-navigation/navigation2 #356: Issues with yaml-cpp-vendor and nav2_dynamic_params

abylikhsanov, 2018-11-23:
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:

I had downloaded yaml-cpp-vendor and built it using colcon build. It worked well on map_server but after sometime, I have decided to install the whole nav2 package and now, whenever I try to build the whole nav2 package I get this error:

```
Could not find a package configuration file provided by "yaml_cpp_vendor"
  with any of the following names:

    yaml_cpp_vendorConfig.cmake
    yaml_cpp_vendor-config.cmake
```
This is so strange as it worked before and know it complains that it can't find the yaml-cpp-vendor. This error is from the nav2_map_server.

Also, I get the following errors (from nav2_dynamic_params):

```
navigation2/nav2_dynamic_params/include/nav2_dynamic_params/dynamic_params_client.hpp:240:29: error: ‘using element_type = class rclcpp::Node {aka class rclcpp::Node}’ has no member named ‘get_node_names’; did you mean ‘get_name’?
     auto node_list = node_->get_node_names();
                             ^~~~~~~~~~~~~~
                             get_name
/home/ikhsanov/ws/navigation2_ws/src/navigation2/nav2_dynamic_params/include/nav2_dynamic_params/dynamic_params_client.hpp:243:35: error: unable to deduce ‘auto&&’ from ‘node_list’
     for (const auto & node_name : node_list) {

```

abylikhsanov, 2018-11-26:
Whoever gets this error check your rclcpp version. After looking at ros2/rclpp I have figured out that the official ROS2 installation tutorial is installing rclcpp "bouncy" branch whereas nav2 packages work using a different rclcpp branch which is quite different.

closed 2018-11-26 by abylikhsanov

mkhansen-intel, 2018-11-26:
@abylikhsanov - yes we're working off the master ROS2 branches for the ros2.repos because there are changes that weren't in the bouncy release that we need, and we're targeting the Crystal release in a few weeks. Sorry for the confusion.