ros-navigation/navigation2 #363: Ros2 navigation2 can not load map in RVIZ2

congliu0913, 2018-11-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - initial_ros_setup.sh
- Version or commit hash:
  - nav2_bringup_readme branch
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
As we follow https://github.com/ros-planning/navigation2/tree/nav2_bringup_readme/nav2_bringup to load map by ros2 navigation, the map not show in rviz2. 
```
$ ros2 launch nav2_bringup nav2_bringup_1st_launch.py map:=/home/intel/Documents/map/cartographer_40hz_default_speed_1.yaml 
```
```
$ ros2 run rviz2 rviz2 
```
We no show map topic in ros2 topic list.
```
$ ros2 topic list 
/amcl_pose
/initialpose
/occ_grid
/parameter_events
/particlecloud
/scan
/tf
/tf_static

$ ros2 node list
ros2 node  list
/launch_ros
/map_server
/amcl
/amcl
/amcl
/rviz
```
#### Expected behavior
Can display map in RVIZ2.
#### Actual behavior
Unable to display map in RVIZ2.
#### Additional information
[image]
[image]
[image]
[image]
[image]

mkhansen-intel, 2018-11-27:
The map is published on the /occ_grid topic since the type is /occ_grid. We have an issue #197 to change this.

congliu0913, 2018-11-28:
After change topic from /map to /occ_grid, we can show loaded map in rviz2.
[image]

closed 2018-11-28 by mkhansen-intel