ros-navigation/navigation2 #365: Building dwb_plugins failed to use initial_ros_setup.sh on Ubuntu 1804

congliu0913, 2018-11-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - initial_ros_setup.sh
- Version or commit hash:
  - master branch
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I failed when I build the environment of navigation2 using initial_ros_setup.sh.
```
./initial_ros_setup.sh
...
Starting >>> dwb_plugins
--- stderr: dwb_plugins
/home/robot/nav2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/test/twist_gen.cpp:38:10: fatal error: gtest/gtest.h: No such file or directory
 #include "gtest/gtest.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/twist_gen_test.dir/test/twist_gen.cpp.o] Error 1
make[1]: *** [CMakeFiles/twist_gen_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< dwb_plugins        [ Exited with code 2 ]
```

#### Expected behavior
The navigation2 environment was build successfully.
#### Actual behavior
Failed while building dwb_plugins.
#### Additional information
When I manually installed libgtest-dev, the build was successful.
```
$ sudo apt install libgtest-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  googletest
The following NEW packages will be installed:
  googletest libgtest-dev
0 upgraded, 2 newly installed, 0 to remove and 50 not upgraded.
Need to get 654 kB of archives.
After this operation, 7,608 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```

mkhansen-intel, 2018-11-27:
@crdelsey - I think this is due to a missing depend on 'ament_cmake_gtest' in the CMakeLists.txt file. Would that fix this?

mkhansen-intel, 2018-11-27:
I had seen this on 16.04 but thought it was a Gtest version problem (#353). Are you on 18.04 as you said in the Operating System field above?

crdelsey, 2018-11-27:
The failure seems to happen on the `rostest` based tests. These should probably use `ament_add_gtest_executable` instead of `add_executable`

Since these tests aren't actually fully ported and don't run, we could just not build them for now.

congliu0913, 2018-11-28:
I installed it on a pure Ubuntu 18.04 and follow the Build Instructions(https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md).
The gtest version I installed is 1.8.
```
$sudo apt search libgtest
Sorting... Done
Full Text Search... Done
libgtest-dev/bionic,now 1.8.0-6 amd64 [residual-config]
  Google's framework for writing C++ tests [source code]
```

PR #424 (2018-12-07): Fix dwb_plugin test build
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | Resolves #353. Resolves #365 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on |  NA  |

---

## Description of contribution in a few bullet points

* Fixed build failures that sometimes occurred in dwb_plugins tests. They are now built as gtest executables instead of just executables.
* The tests no longer require rostest. They are just standard gtest executables and can be launched without a launch file or other support.
* Enabled the goal_che [... cut]
closed 2018-12-07 by crdelsey