ros-navigation/navigation2 #366: How to launch the bringup launch file?

abylikhsanov, 2018-11-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04>

I have deleted the bouncy version of ROS 2 in order to install navigation2. After installing the correct ROS2 version and sourcing the `source ros2_ws/install/local_setup.bash` (together with `/opt/ros/bouncy..`.) I cannot get the ros2 launch command. 

So basically when I type `ros2 launch `+ double tab I do not get anything. 

Am I missing something?

mkhansen-intel, 2018-11-27:
Do you have 'argcomplete' installed? 

You shouldn't need to have /opt/ros/bouncy if you're building ros2 from source.

crdelsey, 2018-11-27:
Does ros2 launch work when you type the full command line?
`ros2 launch nav2_bringup nav2_bringup_launch.py map:=<path to>/map.yaml`

mkhansen-intel, 2018-11-27:
yes, in general ros2 launch works for me, however the argcomplete only works for me for the package name. If you have sourced the setup.bash for the navigation2_ws, it should be able to find your packages.

Does 'ros2 run' argcomplete work for you?

abylikhsanov, 2018-11-28:
Yes, thank you for your help, after installing argcomplete I managed to launch the files in nav2.
Also, I have removed the /opt/ros/bouncy from my bash profile, now I understand more about ROS 2.

Also, how can I stop the process when I launch nav2_bringup? When I try to terminate using Ctrl + C nothing happens. I mean the processes die but I never actually quit.

mkhansen-intel, 2018-11-28:
Unfortunately some of the processes don't respond to Ctrl-C yet. Map server is one we have filed a bug to fix. If this happens, you can Ctr-Z and then list the processes using 'ps'. Then kill the process using 'kill -9 <process id>'

closed 2018-11-29 by abylikhsanov