ros-navigation/navigation2 #370: Robot doesn't stop at goal

mkhansen-intel, 2018-11-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - master at 0892b0762454a73751b86845ded6e31838b633bb
- DDS implementation:
 - Default / Fast RTPS
- Client library (if applicable):
 - rclcpp

#### Steps to reproduce issue
launch everything per the nav2_bringup README

#### Expected behavior
When robot reaches goal, it stops

#### Actual behavior
Robot keeps moving

#### Additional information
This is the relevant log info:
```
[INFO] [amcl]: AmclNode publishing pose
[INFO] [simple_navigator]: Begin navigating to (0.00, 2.00)
[INFO] [navfn_planner]: Attempting to a find path from (-2.00, -0.50) to (0.00, 2.00).
Detected time source change. Clearing TF buffer.[INFO] [world_model]: Received costmap request
[INFO] [navfn_planner]: Making plan from (-2.00,-0.50) to (0.00,2.00)
[INFO] [navfn_planner]: Found valid path of size 145
[INFO] [navfn_planner]: Publishing the valid path.
[INFO] [navfn_planner]: Successfully navigated to (0.00, 2.00) with tolerance 2.00
[INFO] [simple_navigator]: Achieved navigation goal of (0.00, 2.00)
[INFO] [simple_navigator]: Received path of size 145 from planner
[INFO] [simple_navigator]: Sending path to the controller to execute.
Detected time source change. Clearing TF buffer.[INFO] [FollowPathNode]: Starting controller
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libstandard_traj_generator' should be '/libstandard_traj_generator' for better portability
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libsimple_goal_checker' should be '/libsimple_goal_checker' for better portability
[INFO] [FollowPathNode]: DWBLocalPlanner
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libdwb_critics' should be '/libdwb_critics' for better portability
[INFO] [FollowPathNode]: Using critic "RotateToGoal" (dwb_critics::RotateToGoalCritic)
[INFO] [FollowPathNode]: Using critic "Oscillation" (dwb_critics::OscillationCritic)
[INFO] [FollowPathNode]: Using critic "ObstacleFootprint" (dwb_critics::ObstacleFootprintCritic)
[INFO] [FollowPathNode]: Using critic "GoalAlign" (dwb_critics::GoalAlignCritic)
[INFO] [FollowPathNode]: Using critic "PathAlign" (dwb_critics::PathAlignCritic)
[INFO] [FollowPathNode]: Using critic "PathDist" (dwb_critics::PathDistCritic)
[INFO] [FollowPathNode]: Using critic "GoalDist" (dwb_critics::GoalDistCritic)
[INFO] [FollowPathNode]: Initialized
[ERROR] [DWBLocalPlanner]: No valid trajectories out of 399! 
[ERROR] [DWBLocalPlanner]: 1.00: ObstacleFootprint/Trajectory Hits Obstacle.
[INFO] [FollowPathNode]: No valid trajectories out of 399! 
[ERROR] [simple_navigator]: Control task failed.
```
Note the "No valid trajectories" message. That is coming from the illegal trajectory tracker here:
https://github.com/ros-planning/navigation2/blob/0892b0762454a73751b86845ded6e31838b633bb/nav2_dwb_controller/dwb_core/src/illegal_trajectory_tracker.cpp#L69

mkhansen-intel, 2018-11-27:
I've reproduced this twice, and the last time got this value on the /cmd_vel topic:
```
linear:
  x: 0.09578947368421051
  y: 0.00016491353817774806
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.6842105263157895
```

crdelsey, 2018-12-03:
This was resolved by #358. I think we can close this now.

closed 2019-02-15 by mkhansen-intel